bigtreetech/BIGTREETECH-SKR-V1.3 #279: No output on Hot Bed - very low output voltage

Gavin152, 2020-03-29:
So, I finally got around to swapping the stock melzi board of my CR10 for an SKR 1.4 and everything went pretty smoothly so far. However, somehow I can't get the heated bed to work.

The Cr10 is using an external Mosfet and I was panning on continuing to use it. So I'm wiring the bed output on the SKR to the control input of the Mosfet. I'm getting "Heating failed" errors everytime I try to heat the bed. I've narrowed it down so far as I can tell, that the hot bed output on the board is not turning on. I measured the voltage on the board and I only get around 400 mV on the hot bed output (when I turn the bed on) and also the LED next to the Q2 Mosfet is not lighting up.

Could I be simply dealing with a faulty board or is something in my marlin configuration messed up (this is my first time setting up marlin): https://github.com/Gavin152/cr10marlinconf/blob/master/Configuration.h.

Any hints would be greatly appreciated.

Talha909, 2020-04-22:
> So, I finally got around to swapping the stock melzi board of my CR10 for an SKR 1.4 and everything went pretty smoothly so far. However, somehow I can't get the heated bed to work.
>
> The Cr10 is [...]

> So, I finally got around to swapping the stock melzi board of my CR10 for an SKR 1.4 and everything went pretty smoothly so far. However, somehow I can't get the heated bed to work.
>
> The Cr10 is [...]

Did you find any solution. I have the same issue.

Gavin152, 2020-04-22:
> Did you find any solution. I have the same issue.

Yes, damn, meant to post this here anyway. 

I ended up setting the hotend2 output as the bed output. From there I'm using the external MOSFET of the CR10. As long as there's a separate MOSFET in use, there won't be too much current pulled from the output on the board. 

Hope this makes sense.

closed 2020-04-22 by Gavin152

quaint-engineer, 2020-08-10:
This just happened to my SKR 1.4 on my Sidewinder X1. One of my Z steppers started to make a clicking noise also while plugged into E1 also.

gustavomiguelsa, 2021-04-10:
Definitely late to this but did you connect the 'Bed Power In' terminals to your PSU, in addition to connecting the MOSFET signal wire to the 'Hot Bed Out' terminals? Without current flowing to those 'Bed Power In' there's no way the MOSFET signal wire is gonna pass it through to the MOSFET
Leaving this here as future reference for GTR v1.0 or SKR Pro users

jakebullet70, 2024-04-14:
> This just happened to my SKR 1.4 on my Sidewinder X1. One of my Z steppers started to make a clicking noise also while plugged into E1

WOW, Same clicking sound and now dead bed on E1. How did you fix it?