bigtreetech/BIGTREETECH-SKR-V1.3 #281: Bigtreetch skr v1.3 tmc2209

Th0r777, 2020-03-30:
I just bought the skr v1.3 with tmc2209 drivers. I downloaded the firmware adjust everything I needed and every time i compile it I get this all these errors.

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\kqmedia\\Downloads\\BIGTREETECH-SKR-V1.3-master\\BIGTREETECH-SKR-V1.3-master\\BTT SKR V1.3\\firmware\\Marlin-bugfix-2.0.x\\.pio\\libdeps\\LPC1768\\_tmp_installing-242_o68i-package\\TMCStepper-pr_lpctimingfix\\examples\\Calibrate_spreadCycle\\Calibrate_spreadCycle.ino':
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\kqmedia\.platformio\platforms\nxplpc-arduino-lpc176x\builder\main.py", line 148:
    target_elf = env.BuildProgram()
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1035:
    project.install_dependencies()
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 867:
    lm.install(uri)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 280:
    force=force,
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\kqmedia\.platformio\python37\lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\kqmedia\.platformio\python37\lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target:

I can't get past this. Help please?

sketch4x4, 2020-03-30:
is that in visual studio? did you import the marlin folder correctly through PlatformIO or did you just add the folder through file/ open folder?

Th0r777, 2020-03-30:
Yes it is vs code but I also tried atom and got the same thing. I didn’t import the marlin folder at all. I’ve been following a couple of different YouTube videos and they didn’t say anything about importing the marlin folder. How do I go about this?

To tell you the truth I’ve been through a lot of steps but have had a lot of different problem so I thought I’d go back to the original problem and try to fix it correctly.

Th0r777, 2020-04-01:
I changed the original Marlin-bugfix-2.0.x folder name to MarlinFirmware and a new TMC folder in the .pio folder appeared and now I'm getting this error. It's like Git or whatever is creating or downloading the .pio folder isn't getting everything. Do I also need to run a different pyhton version to get the python errors to go away?

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\kqmedia\\Downloads\\BIGTREETECH-SKR-V1.3-master\\BIGTREETECH-SKR-V1.3-master\\BTT SKR V1.3\\firmware\\MarlinFirmware\\.pio\\libdeps\\LPC1768\\_tmp_installing-mnw079e0-package\\Adafruit_NeoPixel-release\\examples\\StrandtestArduinoBLECallback\\StrandtestArduinoBLECallback.ino':
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\kqmedia\.platformio\platforms\nxplpc-arduino-lpc176x\builder\main.py", line 148:
    target_elf = env.BuildProgram()
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\kqmedia\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1035:
    project.install_dependencies()
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 867:
    lm.install(uri)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 280:
    force=force,
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\kqmedia\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\kqmedia\.platformio\python37\lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\kqmedia\.platformio\python37\lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target:

aaltman, 2020-04-05:
This may be an issue with too long of a path, or a path with spaces in it.  Try moving the subfolder from "firmware\MarlinFirmware" somewhere up the tree, like straight into C:\.  It's silly but that resolved a problem that looked similar to this for me.

Th0r777, 2020-04-05:
That worked great but my next problem is I compiled it and now the LCD isn't working. I used a different Marlin that was configured for my Tevo Tarantula and the LCD works with it. I am now trying to find the differences with two, but if anyone knows exactly what I need to change that would be a huge help.

Th0r777, 2020-04-05:
So I found some changes and got the LCD working. Now I'm having TMC connection errors. I have theses errors with the Marlin build too. I found a fix that says I need to add -DHAVE_SW_SERIAL flags to the  build. Now I'm trying to figure where to add those flags in the build. If anyone has an example that would be great.

Th0r777, 2020-04-10:
I got the TMC connection errors to go away, I found I had too many jumpers on under the drivers. Now I am trying to fine tune all the setting. The one I'm having the biggest problems with is homing Z. X and Y work great with sensorless homing. The head goes back to center and just sits there. The Z motor is on because I can't move it by hand. I have tried to adjust the feed rate but it,s not working.