bigtreetech/BIGTREETECH-SKR-V1.3 #284: USB / SD card / TFT module / SKR communication issues and questions

bogus105, 2020-04-01:
Guys!
My setup:
Marlin 2.0 bugfix based firmware + Bigtreetech SKR 1.3 controller board (LPC1768 processor) + 2.4" TFT controller module from Bigtreetech (TFT24-V1.1).
-------------------------
The facts:
SKR 1.3 controller board has:

    USB connector to connect with PC for printing from host (PC), upgrading firmware (sending firmware.bin from PC to SKR over USB cable, then reboot SKR), using as SD card reader (when connected with PC, SKR acts like a SD card reader so we can see what we have on the SD card inserted into the SKR's onboard slot).
    SD card slot - as in 1 we can use it for firmware flashing, as SD card reader or to store print files we want to print in stand-alone mode

TFT24-V1.1 controller has:
3. SD card slot (big version) for flashing the firmware and storing print files and print from it (?) in standalone mode (?)
4. USB port for USB sticks - for storing print files and print from it (?)
------------------------------------

In Vision Studio PlatformIO we can see in "Configuration.h" and "Configuration_adv.h" files some settings for SDSUPPORT for serial communication etc...

The questions i have:

    in "Configuration.h" there is #define SERIAL_PORT -1 and #define SERIAL_PORT_2 0. What these ports really are? What particular ports are responsible for? What the available arguments ([-1, 0, 1, 2, 3, 4, 5, 6, 7]) mean exactly?

    BAUDRATE - i heard i have to set it to 115200. Is it really the only option? During the print (so far only using PC as host over USB cable!) i can see very short (like miliseconds) stops resulting in small blobs all over my print. It looks like USB data transfer problem (Win10 USB problem?) - maybe the USB serial communication buffer shortage?

    If i change BAUDRATE, where do i have to change the setting as well? In my host software? I'm using CURA 4.5. Is this setting (BAUDRATE) responsible for PC (host) - SKR communication ONLY??

    How about communication between SKR and TFT module? I can see there is TX0/RX0 assigned for it on the SKR circuit board. Does it mean TFT module communicates with SKR via SKR's hardware serial 0? If yes is it the same serial as defined in "Configuration.h" file as "#define SERIAL_PORT"? Should it be -1, 0, 1 or what?

Overall design questions:
Correct me if i'm wrong. I uderstand it all may work like this: SKR can talk to onboard SD slot via hardware SPI (MISO, MOSI, SCK and CS pins). So it is a really FAST connection, actually the fastest one in the whole setup. The best one to use as main communication mean for PRINTING. So i can see it as it is the best option to use SKR onboard SD slot for print files in terms of both reliability and speed.
The next way to print something is to use TFT24-V1.1 module which has both SD card slot and USB port for USB sticks. HOWEVER as TFT module communicates with SKR via UARTs RX0/TX0 lines only it has inherent buffer and speed limitations. I do not know the exact fdesign but i assume that during the printing TFT module continouslly is sending G-codes from SD card/USB stick connected to TFT slots to SKR, and SKR is doing all the timing and pulses creation work. If womething happen between TFT module and SKR during the print - the print is ruined. In my opinion it is not the best option to use TFT module slots for print files storage and ANY of the comm problems between the two systems (one is SKR and the second one is TFT processor and its firmware) lead to print failure.
The last way for rinting i'm assuming now is printing from host - Win10 based PC in my case over the USB cable. The same drawbacks as in previous way - weak comm stability and speed in general.

So i'm chosing the following approach:

1. slice the file in Cura1
2. send the g-code file from Cura directly to SD card inserted into the SKR montherboard slot using USB cable connection between PC and SKR (SKR acts like a regular SD card reader).
3. go to TFT touch screen mode, pick the file stored on the SD card on the SKR board and hit "PRINT".

The problem now is when i go to touchsceen and go for PRINT -> OnboardSD i can see "Loading..." for a while and then a message: "Read onboard SD card error".
When i go to: MENU -> Custom -> Init SD card i get echo reply: "SD init fail ok".
But i do not know WHICH SD card we are talking about here? The one in TFT module or in the SKR?

I'm trying to figure out how to set everything up so i can easily use SKR onboard SD slot and controll my printer over USB as well.

Please lets start the discussion here and clarify the problems related to SD card, USBs and communication issue between SKR, TFT modules and hosts.

sorren4, 2020-04-06:
I can't even get my PC or Octo/AstroPrint to communicate with the SKR 1.3/1.4 over USB.  It shows up but can't find a driver or establish a comm link with the PC or Pi. I have the settings match what's above but to no avail. Drivers I found on the site don't seem to work as valid drivers either (Win10 x64).  No one else seems to have an answer yet and/or has the same problem.

bogus105, 2020-04-07:
i don't know. Mine Win10 can see SKR. However i might have the necessary drivers as i'm using Arduino IDE.

HansaG, 2023-03-15:
> 

it's been 3 years, but can I ask if anyone had helped you on this? if so, can I know what exactly you did to overcome this?