bigtreetech/BIGTREETECH-SKR-V1.3 #285: SKR 1.4 Turbo Endstops not responding.

Defaultsound, 2020-04-02:
Hi,

So I'm currently trying to replace my Creality CR10 S5's motherboard with a SKR 1.4 Turbo. I purchased this along with TMC 2208 v3 Drivers. So far I've got most things working, from the motors to temperature sensors etc. However the endstops just won't communicate with the board. 

I've got X,Y,Z endstops all plugged in. Using 1.29 for X, 1.28 for Y and 1.27 for Z. Pressing the endstops and sending M119 through Pronterface does not display or show a change in its state. I haven't the slightest clue to why this is happening. I've check the switches for continuity and they all work. So I'm left to believe I'm missing something in the Configuration.h files.

Any help would be greatly appreciated!

# Configuration.h 

```
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
[... 5104 lines cut ...]
/**
 * Advanced Print Counter settings
 */
#if ENABLED(PRINTCOUNTER)
#define SERVICE_WARNING_BUZZES 3
// Activate up to 3 service interval watchdogs
//#define SERVICE_NAME_1      "Service S"
//#define SERVICE_INTERVAL_1  100 // print hours
//#define SERVICE_NAME_2      "Service L"
//#define SERVICE_INTERVAL_2  200 // print hours
//#define SERVICE_NAME_3      "Service 3"
//#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

/**
 * M43 - display pin status, watch pins for changes, watch endstops & toggle LED, Z servo probe test, toggle pins
 */
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

```

Defaultsound, 2020-04-03:
Solved this issue. Turns out that the positions of the JST connectors were wrong. On the switch itself the pins are on the out most side of the connector. As the original motherboard used a 2 pin connector I had to replace them with 3 pin connectors for the SKR. However, putting them on the far most pins does not work. Instead you need to put them on the far most right side of the connector on the JST. Here is a picture to help demonstrate. https://imgur.com/gallery/bMtPcU0

closed 2020-04-03 by Defaultsound

ypsonar, 2020-09-08:
Hi, Please help
if you are using mechanical endstop, then Please kindly share latest your updated marline 2.0.