bigtreetech/BIGTREETECH-SKR-V1.3 #290: TMC5160 SKR 1.3->1.4 turbo : Testing X connection ALL HIGH

superkefir, 2020-04-11:
Hello, i have a problem with my new 5160 v1.2 drivers.
I got three boards: 2 pcs SKR 1.3 and one SKR 1.4 Turbo. 
I have Marlin 2.0.5.2 - > 2.0.5.3 versions
When i turn off the board and send M122 i see:
ПОЛУЧЕНО: Driver registers:
ПОЛУЧЕНО: 		X	0x80:06:40:00
ПОЛУЧЕНО: 		Y	0x80:07:40:00
ПОЛУЧЕНО: 		Z	0x80:04:40:00
ПОЛУЧЕНО: 		Z2	0x80:04:40:00
ПОЛУЧЕНО: 		E	0x80:04:40:00
Testing X (and all others) connection... OK

Also i can see M913 result of threshold.

But when i start the print during the start g-code i send M122 and i the result is always: Testing X connection ALL HIGH on all drivers. When i stop the print it's the same. Untill i reboot the board. Then everything goes in a circle.

Is it wrong connection of wires or smth else?
I can't find the solution to this problem for three days by google, so maybe you can help me.

superkefir, 2020-04-11:
Little update.
I check diag pin and turn it off, i turn off sensorless homing. 
Then i checked G-code after what line it's bugged.
So if i turn printer then send M122 it's ok and shows registers and connection ok.
Then i send M109 S220 T0
And printer echo busy. At this moment i send M122.
And right after busy lines when the temperature get's stabilized i get the error by all drivers and no connection.

closed 2020-04-15 by superkefir

Mace-maker, 2020-06-06:
I have the same problem 1.4t did you find a solution?

superkefir, 2020-06-06:
Yes i unbind the pins. TMC_SW_SCK and RGB_LED_G_PIN had the same pin so after starting the print the pin of led blocked the pin of TMC_SW_SCK. So i have to unbind it.
This one:
  #define RGB_LED_G_PIN 4

Mace-maker, 2020-06-06:
Ok, thx, I don’t think I have led uncommented, but I’ll check it tomorrow. Pins file and Configuration.h?

superkefir, 2020-06-07:
Yes, also you can check M43 output to find if there any pins are double used.

Mace-maker, 2020-06-07:
I don’t know exactly what I’m looking for, but I can’t see pin numbers that are assigned to two variables at the same time, maybe the empty “.”s without numbers like E1_DIAG_PIN mean that they’re not set properly? Only diag I can see are on the drivers themselves?
[image]
[image]

superkefir, 2020-06-08:
You can compare with my output log.
ПОЛУЧЕНО: PIN: 0.00        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.01        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.02 (A 2)  TXD0                                   Input  = 1
ПОЛУЧЕНО: PIN: 0.03 (A 3)  RXD0                                   Input  = 1
ПОЛУЧЕНО: PIN: 0.04        TMC_SW_SCK                             Output = 1
ПОЛУЧЕНО: PIN: 0.05        TMC_SW_MISO                            Input  = 1
ПОЛУЧЕНО: PIN: 0.06        <unused/unknown>                       Output = 1                       
ПОЛУЧЕНО: PIN: 0.07        <unused/unknown>                       Input  = 0                       
ПОЛУЧЕНО: PIN: 0.08        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.09        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.10        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.11        E0_DIR_PIN                             protected 
ПОЛУЧЕНО: PIN: 0.15        SCK_PIN                                Input  = 1
ПОЛУЧЕНО: PIN: 0.16        SDSS                                   Analog in =     0
ПОЛУЧЕНО: .                SS_PIN                                 Output = 1
ПОЛУЧЕНО: PIN: 0.17        MISO_PIN                               Input  = 1
ПОЛУЧЕНО: PIN: 0.18        MOSI_PIN                               Input  = 1
ПОЛУЧЕНО: PIN: 0.19        Y_STEP_PIN                             protected 
ПОЛУЧЕНО: PIN: 0.20        Y_DIR_PIN                              protected 
ПОЛУЧЕНО: PIN: 0.21        Z_ENABLE_PIN                           protected 
ПОЛУЧЕНО: PIN: 0.22        Z_STEP_PIN                             protected 
ПОЛУЧЕНО: PIN: 0.23 (A23)  <unused/unknown>   Analog in =     0   Input  = 0                       
ПОЛУЧЕНО: PIN: 0.24 (A24)  <unused/unknown>   Analog in =   975   Input  = 0                       
ПОЛУЧЕНО: PIN: 0.25 (A25)  <unused/unknown>   Analog in =   974   Input  = 0                       
ПОЛУЧЕНО: PIN: 0.26 (A26)  <unused/unknown>   Analog in =   823   Input  = 0                       
ПОЛУЧЕНО: PIN: 0.27        <unused/unknown>                       Input  = 0                       
ПОЛУЧЕНО: PIN: 0.28        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.29        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 0.30        <unused/unknown>                       Input  = 0                       
ПОЛУЧЕНО: PIN: 1.00        POWER_LOSS_PIN                         Input  = 1
ПОЛУЧЕНО: PIN: 1.01        E1_CS_PIN                              protected 
ПОЛУЧЕНО: .                Z2_CS_PIN                              protected 
ПОЛУЧЕНО: PIN: 1.04        E0_CS_PIN                              protected 
ПОЛУЧЕНО: PIN: 1.08        Z_CS_PIN                               protected 
ПОЛУЧЕНО: PIN: 1.09        Y_CS_PIN                               protected 
ПОЛУЧЕНО: PIN: 1.10        X_CS_PIN                               protected 
ПОЛУЧЕНО: PIN: 1.14        E1_DIR_PIN                             protected 
ПОЛУЧЕНО: .                Z2_DIR_PIN                             protected 
ПОЛУЧЕНО: PIN: 1.15        E1_STEP_PIN                            protected 
ПОЛУЧЕНО: .                Z2_STEP_PIN                            protected 
ПОЛУЧЕНО: PIN: 1.16        E1_ENABLE_PIN                          protected 
ПОЛУЧЕНО: .                Z2_ENABLE_PIN                          protected 
ПОЛУЧЕНО: PIN: 1.17        TMC_SW_MOSI                            Output = 0
ПОЛУЧЕНО: PIN: 1.18        <unused/unknown>                       Output = 1                       
ПОЛУЧЕНО: PIN: 1.19        <unused/unknown>                       Output = 1                       
ПОЛУЧЕНО: PIN: 1.20        <unused/unknown>                       Output = 1                       
ПОЛУЧЕНО: PIN: 1.21        <unused/unknown>                       Output = 1                       
ПОЛУЧЕНО: PIN: 1.22        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 1.23        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 1.24        NEOPIXEL_PIN                           Input  = 0
ПОЛУЧЕНО: PIN: 1.25        FIL_RUNOUT2_PIN                        Input  = 1
ПОЛУЧЕНО: .                E1_DIAG_PIN                            Input  = 1
ПОЛУЧЕНО: PIN: 1.26        E0_DIAG_PIN                            Input  = 1
ПОЛУЧЕНО: PIN: 1.27        Z_MIN_PIN                              protected 
ПОЛУЧЕНО: .                Z_DIAG_PIN                             protected 
ПОЛУЧЕНО: .                Z_STOP_PIN                             protected 
ПОЛУЧЕНО: PIN: 1.28        Y_MIN_PIN                              protected 
ПОЛУЧЕНО: .                Y_DIAG_PIN                             protected 
ПОЛУЧЕНО: .                Y_STOP_PIN                             protected 
ПОЛУЧЕНО: PIN: 1.29        X_MIN_PIN                              protected 
ПОЛУЧЕНО: .                X_DIAG_PIN                             protected 
ПОЛУЧЕНО: .                X_STOP_PIN                             protected 
ПОЛУЧЕНО: PIN: 1.30 (A62)  <unused/unknown>   Analog in =   757   Input  = 0                       
ПОЛУЧЕНО: PIN: 1.31 (A63)  <unused/unknown>   Analog in =   741   Input  = 0                       
[... cut ...]
ПОЛУЧЕНО: PIN: 2.13        E0_STEP_PIN                            protected 
ПОЛУЧЕНО: PIN: 3.25        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 3.26        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 4.28        <unused/unknown>                       Input  = 1                       
ПОЛУЧЕНО: PIN: 4.29        <unused/unknown>                       Input  = 1                       

It seems like some pins are used twice. What configuration you use for your printer?

Mace-maker, 2020-06-08:
Yeah, looks like where I have empty points that might mean double use like filament run out, I’m using kay3d cheetah 5.0 firmware, I have a bl touch and filament run out sensor. Parallel lead screws so 2 Zs,  nothing special.
The TFT 35 is working, but I see points there too.
Thanks for the file, I’ll investigate end of the week after work.