bigtreetech/BIGTREETECH-SKR-V1.3 #292: TFT 24 V1.1 not connecting to SKR V 1.4 Turbo

j3cc, 2020-04-12:
I do not get a working connection to my tft 24 V1.1. I successfully updated the firmware of both parts. I flashed the tft from firmware version 1.23 to the latest (1.26) which I downloaded as binary and put on the sd-card. Update was successful. 
I also compiled the latest marlin 2.0 with platform.io and successfully loaded it onto the SKR 1.4 Turbo board. I can see an external drive with the firmware on, if I plug the motherboard in. With pronterface however, I can't get a connection running. It correctly recognizes the external device, but no matter what baud raute i put in, no connection to printer this way possible.

The tft24 does also not connect to the skr 1.4 turbo board. It just says "No printer attached". No matter what I enter as connection rate, it does not work.
I already plugged out the thermistors of the printer because I've read that there may be some problems with them but to no avail.
How do I go about debugging this issue? Are there some light blinking codes or something that indicate the skr 1.4 turbo is working as intended?

Thx in advance

j3cc, 2020-04-12:
Ok, I finally figured it out. The screen has to be connected via RS232 interface. If it is only connected via EXP1 and EXP2, it will not connect to the board.

closed 2020-04-12 by j3cc

RavenDwn, 2020-06-17:
I have the same issue hooked up on rs232 but still can not get it to attach printer. I thought there was a post or video to change config but can't find it.  
Any thought on how the get it to read board? All firmware has been updated with success except printer