bigtreetech/BIGTREETECH-SKR-V1.3 #294: X and Y motors stop, while extruder and Z keep going. Help Please.

Demxsr, 2020-04-14:
I just built a new CoreXY printer using an SKR 1.3 and have been having a recurring issue where the X and Y motor stop moving, mid print, at the same time. This happens at random points during a print. Not at a particular time, layer height, etc. Many times it is 8-10 hours into a print that is going perfectly up to that point. But it has also happened faster. There are no errors displayed on the screen when it happens. The extruder continues extruding and retracting and the dual Z motors also keep stepping, as if the print is continuing as normally. The hotend and heated bed also maintain their temperatures. I’ve had some prints complete successfully, but most of them don’t, due to this issue. And it seems the longer the print, that it’s only a matter of time till this happens. This leads to a massive blob of filament forming around my hotend. Not fun, and maybe dangerous?

[image]

[image]

This is the hardware I’m running:
SKR 1.3
Nema 17, 1.7A 40Ncm Steppers X5
Eryone TMC2208 V1.2
Bigtreetech TFT35 V3.0

Here's the firmware:
[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4474345/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4474347/Configuration_adv.h.txt)

I’ve bridged the pads on my 5 stepper drivers (X, Y, Z, Z2, and E0) to run them in UART and have also placed the 5 “shunts” on the correct areas of the board to have UART enabled. 

I have heatsyncs on all the stepper drivers with a fan blowing directly across them. When the problem occurs the heatsyncs are barely warm to the touch. 
The motors are also barely warm to the touch when the problem occurs. 
I’ve tried multiple SD cards, and formatting them.
I’ve re-wired all the connectors to the board.
I’ve tried printing from the SD slot on the screen and the onboard SD slot.

The board isn’t recognized when plugged in via USB to my laptop anymore. Even though the printer can still connect via Pronterface with USB and I can execute movements and commands through it. I had this issue before, and it fixed itself after fiddling with it for a few hours. But now it seems to be permanently unrecognized. I was able to upload new firmware builds via USB and SD cards no problem before. I’ve been having the stop print issue from day 1, so I’ve done multiple flashes in an attempt to isolate the issue. When I try to upload a new firmware build now, I get a “Error: please specify upload port.” So I try to update my firmware with an SD card using the .bin file. When I power cycle the board, the .bin goes away leaving only the .cur. So it looks like it’s updating the firmware. But if I run M122 on pronterface the electrical current settings for the motors are all set to 600 when I’ve tried to change those figures up or down to see if that was causing my problem. So it appears as if the firmware is getting flashed based on the file, but not in realty?

The only error I get on the screen is after I tell the print to stop printing, once I see that this problem has occured. After a few seconds of the print head not moving I get “Homing Failed: PRINTER HALTED Please Reset.” I’m assuming this is from the printer trying to execute the G1 command at the end of my prints and seeing that the end stops aren’t being triggered after a certain amount of time?

I’m beginning to think the board might have an issue. As I’ve tried to read into this it seems the USB connection issue has happened to many people. Some have encountered some circuits that break after using the board a bit. Unfortunately, I currently lack the knowledge on how to diagnose these types of issues on a technical level.

Sorry for the novel, but I’m just trying to cover all the variables. As this has been a huge time sink trying to figure this out. Many failed prints, after many hours of them going perfectly up to that point. Any advice would be greatly appreciated and any questions welcomed. Thanks

Demxsr, 2020-04-26:
I ended up fixing this problem by replacing the board. I also added a MOSFET for the bed. So it appears that the issue was a defective board. What was wrong with the board? I couldn't tell you, but just replacing the entire board solved weeks of headaches.

miran985, 2020-04-29:
HI!
I have exactly the same problem.
so i must change board? i also have problem with uart mode with 2208 uart.

Demxsr, 2020-04-30:
> HI!
> I have exactly the same problem.
> so i must change board? i also have problem with uart mode with 2208 uart.

I would return it and replace it. That is what fixed my issue. Keep in mind, I also re-did all my wiring to make sure it wasn't a bad connection somewhere. This was prior to me replacing the entire board, in an attempt to isolate the problem.  If BOTH your X and Y motors stop at the same time and the motors and drivers don't feel hot, to the point that they could be over heating, I would assume the issue is with the board. Like what I had.

In my case, I'm also running a CR10S heated bed. Which I've read to be 200 watts, and on here I've read that the board manufacture recommends that you don't do a heated bed of more than 144 watts. So not having the MOSFET for the bed (which I added at the same time as the new board) might help take that big load off the board. Whether that added to or caused the problem, I'm not sure. But for $14, it was worth the peace of mind for me.

Since I've replaced the board and added the MOSFET none of my prints have failed. from 40 minute prints to 2 day prints.

Are you getting the TMC Error on the screen with your 2208's?

Let us know what the outcome is.

miran985, 2020-04-30:
HI!
I change today board.
Now uart works and all seems to be ok...
yes i have the same problem as you write. x or y stop sudenly.
no heat or no bad connection. When restart printer all work.

now i testing second board. seems to be much better.