bigtreetech/BIGTREETECH-SKR-V1.3 #297: SKR 1.4 Turbo not able to init SD card, unable to update firmware.

kulmosen, 2020-04-18:
I have bought the new SKR 1.4 Turbo, and after installing all the cables and drivers (TMC 2209) I get "TMC CONNECTION ERROR". I try to compile a new version of the Bigtree Tech Marlin 2.0.1, with all the correct settings. 
But the SD card reader on the controller is not initialized. I can contact the board with pronterface (now Printrun), and move all the axis manually, I can heat up the bed and hot-end also manually. But I cant send any g-code file to the board, then I get the error "Exception in thread Thread-8:"
I need some help in getting the correct firmware on the board, with the SD card reader enabled. Plus I need to solve the TMC connection error. Can anyone help me?

Atlas-39, 2020-04-20:
I've had the "TMC CONNECTION ERROR" pop up too. It's highly likely that one or more of your drivers are bad. Try sending M122 if you are able to connect it to anything and it will send you stepper driver diagnostics.
As for the SD card reader not working, I'm not too sure, there might be something wrong with the hardware. (Maybe double check your wiring?).
Best of luck

kulmosen, 2023-04-05:
So to update this after a long time. "TMC CONNECTION ERROR" was because I installed a TMC driver that didn't get used for anything. So if anyone bump into this, check if a stepper motor is connected to your TMC driver.