bigtreetech/BIGTREETECH-SKR-V1.3 #300: Unable to flash SKR 1.3

Di3mex, 2020-04-19:
My SKR 1.3 isn't updating itself from SD card anymore. I put firmware.bin onto the sd card on the SKR 1.3 and restart it. Either by using the reset button or the power cycling it. Then I insert the sd card into my pc and find that firmware.bin hasn't been renamed to firmware.cur.

I have also tried using an ST-Link V2 adapter connected to the ICSP header to flash the board. I have connected pins GND, 3V, CLK, DIO and NRST. I am able to connect to the LP1768 using Openocd.

1. Starting OpenOCD

` bin\openocd.exe -f scripts\interface\stlink.cfg -c "reset_config none" -f scripts\target\my_lpc17xx.cfg`

> xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> none separate
> 
> Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : clock speed 10 kHz
> Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
> Info : Target voltage: 3.174396
> Info : lpc17xx.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'telnet' connection on tcp/4444

2. Flashing via OpenOCD

`program N:/D/3D_printing/firmware/Marlin-2.0.x_Mendel90/.pio/build/LPC1768_openocd/firmware.elf; verify; reset`

```
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x1fff0080 msp: 0x10001ffc
** Programming Started **
Verification will fail since checksum in image (0x00000034) to be written to flash is different from calculated vector checksum (0xb98911ec).
To remove this warning modify build tools on developer PC to inject correct LPC vector checksum.
error writing to flash at address 0x00000000 at offset 0x00000000
embedded:startup.tcl:449: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 508
at file "embedded:startup.tcl", line 449
jtag status contains invalid mode value - communication failure
Polling target lpc17xx.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Previous state query failed, trying to reconnect
```

3. Debug output

Here is the output of a few commands executed via OpenOCD.

> reset halt
> target halted due to debug-request, current mode: Thread
> xPSR: 0x01000000 pc: 0x1fff0080 msp: 0x10001ffc
>
> flash list
> {name lpc2000 base 0 size 0 bus_width 0 chip_width 0}
> 
> flash banks
> #0 : lpc17xx.flash (lpc2000) at 0x00000000, size 0x00000000, buswidth 0, chipwidth 0
> 
> flash info 0
> Flash protection check is not implemented.
> #0 : lpc2000 at 0x00000000, size 0x00080000, buswidth 0, chipwidth 0
>         #  0: 0x00000000 (0x1000 4kB) protected
>         #  1: 0x00001000 (0x1000 4kB) protected
>         #  2: 0x00002000 (0x1000 4kB) protected
>         #  3: 0x00003000 (0x1000 4kB) protected
>         #  4: 0x00004000 (0x1000 4kB) protected
>         #  5: 0x00005000 (0x1000 4kB) protected
>         #  6: 0x00006000 (0x1000 4kB) protected
>         #  7: 0x00007000 (0x1000 4kB) protected
>         #  8: 0x00008000 (0x1000 4kB) protected
>         #  9: 0x00009000 (0x1000 4kB) protected
>         # 10: 0x0000a000 (0x1000 4kB) protected
>         # 11: 0x0000b000 (0x1000 4kB) protected
>         # 12: 0x0000c000 (0x1000 4kB) protected
>         # 13: 0x0000d000 (0x1000 4kB) protected
>         # 14: 0x0000e000 (0x1000 4kB) protected
>         # 15: 0x0000f000 (0x1000 4kB) protected
>         # 16: 0x00010000 (0x8000 32kB) protected
>         # 17: 0x00018000 (0x8000 32kB) protected
>         # 18: 0x00020000 (0x8000 32kB) protected
>         # 19: 0x00028000 (0x8000 32kB) protected
>         # 20: 0x00030000 (0x8000 32kB) protected
>         # 21: 0x00038000 (0x8000 32kB) protected
>         # 22: 0x00040000 (0x8000 32kB) protected
>         # 23: 0x00048000 (0x8000 32kB) protected
>         # 24: 0x00050000 (0x8000 32kB) protected
>         # 25: 0x00058000 (0x8000 32kB) protected
>         # 26: 0x00060000 (0x8000 32kB) protected
>         # 27: 0x00068000 (0x8000 32kB) protected
>         # 28: 0x00070000 (0x8000 32kB) protected
>         # 29: 0x00078000 (0x8000 32kB) protected
> lpc2000 flash driver variant: 2, clk: 4000kHz

At this point I don't really care how to flash the board. I just want to somehow be able to flash it. I have a variety of other serial converter chips which I haven't used yet. I am not sure what the best chip for this board is though. 
* ST-LINK V2
* Arduino configured as ArduinoISP
* FTDI232H
* ESP-Prog board

I have wasted countless hours on this already and really hope someone can help me or point me in the right direction.

Marsman1970, 2020-04-19:
Here is a thread  i started many people have posted & helped.
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/166
Here is the Boot loader I ripped of my skr 1.3 board i needed it because i killed a board.
I used a ESP-Prog board. Hope this helps.

I got mine to connect to the pc but not mass storage. If you want to know about loading a boot loader it's a bit of a procedure, it took me a while to figure it out. There is 2 pins under the Z axis you need to short under the Z axis which says boot so the usb serial works it's a bit tricky though. The way i got it to work i used a screw driver to short the pins so you press the reset & short the boot pins then release reset & very quickly unshort boot pins. It took me a while until Flash Magic could read the chip.

Cheers Michael.

Di3mex, 2020-04-20:
I can connect to FlashmagicTool and I also successfully flashed the Bootloader. However, the SD update mechanism still doesn't work. Can I also flash the firmware with FlashMagic?

Marsman1970, 2020-04-23:
If you plug in the usb from the skr 1.3 into the pc can you see the sd card on the skr board.
Michael