bigtreetech/BIGTREETECH-SKR-V1.3 #302: Suddenly boots up with MAXTEMP HEATBED and no USB connection

bywebberen, 2020-04-22:
This is soooo alarming, I have been printing for months with this board on my DADSON PRUSA i3 - Today I turned it on on the speaker started beeping with the error "maxtemp error" heatbed.

 Then I thought - I'll just reflash the marlin (since it has been working fine) -only to find that there is not a USB connection anymore when i connect to my computer. HMMMM... Okay so I pull the sd card and put the firmware directly to the sd card from platform.io - reinserted and did a reset (like always) still error persists, and no USB still. But if i take the SD-card out it will show that the firmware was indeed updated.  . .

WTF??

Alaming because I actually odered a bunch og these SKR v1.3 cards to upgrade my other printers.

bywebberen, 2020-04-22:
UPDATE:
If I disable the "heatbed" in marlin sensor statement 0 instead of 1 it will power on and print. And the USB connection works again...

Marsman1970, 2020-04-23:
Can you check the sensor Maybe you have broken wires to the sensor.
Michael

bywebberen, 2020-04-23:
I did that already, I replaced it with a new one, same result... 🤔

Venlig hilsen / best regards
Niels Østbjerg
Syrenvænget 8
5792 Årslev

+45 40630201

bywebberen, 2020-04-23:
Also if I set the sensor to 0 in marlin, the printer will run, the USB will
work, but of course with no heated bed..

Venlig hilsen / best regards
Niels Østbjerg
Syrenvænget 8
5792 Årslev

+45 40630201

tor. 23. apr. 2020 14.48 skrev Niels Østbjerg <webberen2502@gmail.com>:

bywebberen, 2020-04-25:
Okay, so this time I fired it up, now it does not work at all! 12v goes to the fan through the board, but nothing else works... 

Crap!

bywebberen, 2020-04-25:
flashed my marlin compile to another SKR v1.3 board - it works - too bad the faulted board only worked a couple of months, that could end up being expensive...

I would love to hear some ideas how to trobleshoot it - and fix it ...

bywebberen, 2020-04-26:
all of a sudden the new SKR v1.3 board dropped the heatbed saying -14 degrees C and starts beeping away???

I have only tested and made firmware changes ?

Any ideas ?

bywebberen, 2020-04-26:
> Can you check the sensor Maybe you have broken wires to the sensor.
> Michael

Still the sensor is fine...

bywebberen, 2020-04-26:
And with this reset... now it suddenly jumped to 16 degrees, while the hotend stays at roomtemp 23

closed 2020-04-26 by bywebberen

bywebberen, 2020-04-26:
Might have been a broken wire after all...

Marsman1970, 2020-04-27:
Good to hear you sorted it.