bigtreetech/BIGTREETECH-SKR-V1.3 #304: Incorrect motor movement and no homing

b3nnb, 2020-04-25:
I just purchased a BTT SKR 1.3 and BTT TMC2209v1.2 drivers.
i am installing it into a Kossel mini that i previously had an 8 bit with TMC2208 working just fine.

i'm running the firmware located in this BTT repo. enabled sensorless homing and the TMC_DEBUG. the drivers are being detected in M122.
A M119 shows that all axis endstops are all triggered (not sure if thats normal with sensorless homing)
I have the jumpers on the board for each axis for UART mode and on the endstop jumpers

If i hit home, nothing happens. If i move a single axis, all 3 towers move in the right direction but in a very incorrect amount or not at all.

i also used the same thermistor number as i did on the previous board in the kossel but im getting a -14 degrees

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4532586/Configuration.zip)

b3nnb, 2020-04-26:
i got the motor movement sorted out. had to put Y inverted and somehow the orientation if the front of the printer has changed, no biggie on that. 

but sensorless homing still does not work. M119 still has all as TRIGGERED. i tried homing sensativity from 1-255 at 50pt intervals and not one of them did anything. i do feel the motor engage  when trying to home it, but no movement

cheerynik, 2020-05-02:
skr 1.3 tmc 2208 delta printer. 

axis movement problems occurred when replacing the motherboard. the following configuration solved all the problems and the delta is now working fine. try changing the X axis instead of the Y axis

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR false
#define INVERT_Y_DIR true
#define INVERT_Z_DIR true