bigtreetech/BIGTREETECH-SKR-V1.3 #307: TMC220X UART Serial Port Conflict

rigg77, 2020-04-26:
### Hardware/Software Details:

- **Board:** SKR 1.4 Turbo, **Driver:** TMC 2209 (UART), **SD:** Samsung (ie, not stock)
- **OS:** Win10/Ubuntu/Octopi, **FW:** Main Stream Marlin 2.0.5.3, **IDE/Compiler:** VSCode/Platformio, Atom/Platformio

### Issue Description:

- FW Compiles and uploads perfectly (running headless, confirmed by NEOPix and BLTouch boot appropriately)
- Connection to Pronterface/Octoprint terminals: `#define *_DRIVER_TYPE TMC2209` creates serial port conflict, debug returns 
> serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

### Troubleshooting:

- Built FW w/ `#define *_HARDWARE_SERIAL  Serial1` uncommented 
**Result:** No change. --SIDE NOTE: pins_BTT_SKR_1_4.h line 104, should `X_HARDWARE_SERIAL` use Serial1 like the rest of the defined HW Serial? Why is it using Serial and not Serial1?

- Rebuilt FW as `*_DRIVER_TYPE TMC2209_STANDALONE`, serial port conflict disappears
**Result:**
> Connecting...
echo:SD card ok
echo:EEPROM version mismatch (EEPROM=? Marlin=V76)
echo:Hardcoded Default Settings Loaded
Error:Printer halted. kill() called!

rigg77, 2020-04-27:
Wild day of compiling and uploading, starting from scratch, and working backwards, forwards, and sideways...

Renarrowed issue to EEPROM Settings

Issue resolved with workaround provided by @DoctorDanke in [SKR Mini E3 Issue 127](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/127#issuecomment-560258529), added `#define FLASH_EEPROM_EMULATION` in Configuration.h

Issue still open until maintainer can provide a proscribed solution.

Drzet, 2020-08-28:
But your TMC's are still in standalone mode? Means no sensorless homing.
Strugling with the same issue for 3 days now

rigg77, 2020-08-30:
> But your TMC's are still in standalone mode? Means no sensorless homing.
> Struggling with the same issue for 3 days now

Almost 6 months ago, I don't recall what ended up working. I did get UART running, and everything working. I remember that the FLASH_EEPROM_EMULATION setting corrected a lot of issues for me. My branch and config files compile and upload fine last I checked, recently sent them to a friend for his board upgrade and it worked fine.

End of the day, I see Flash EEPROM is a workaround, not really a solution.