bigtreetech/BIGTREETECH-SKR-V1.3 #309: Bug SKR1.4 TURBO Homing

dono89550, 2020-05-01:
hi everybody,

Since yesterday I have been in total incomprehension.

I recently installed a SKR1.4 turbo with TMC2209 and I compiled marlin 2.0.5.3 and everything worked fine until yesterday.

Print in progress almost finished and there my print comes off the bed (error on my part during the preparation). I stop the print via the screen and I turn off the printer it was 11:30 p.m.

In short, this morning, I turn it back on to launch a new print and impossible to make a home correctly without the X-carriage going two times to the left as if to go to the endstop and stop ... the screen tells me a "printer halted. kill () called" error with no other information ...

I tested my drivers = OK

I tested my Vref = OK

Reflash the motherboard = do not solve the problem

Motor test = OK

I'm pulling my hair out ... if anyone has an idea it would be great !!!

Thank you for your help.

zirrer, 2020-05-26:
Can you confirm that endstops are triggered? 
You can run a M119 to see status.
Run M119 to see current state, then press and hold the suspected endstop with finger, stick, or rubber band and run m119 again that end stop status should change to triggered if it dosent check connection for endstop . Just a thing to check.
I hope this hope or you already got it fixed

samueldenismotta, 2020-06-17:
Check thermistor hotend.

samueldenismotta, 2020-06-17:
Check thermistor hotend