bigtreetech/BIGTREETECH-SKR-V1.3 #310: SKR 1.4 Turbo Wont compile

JBTechRacing, 2020-05-02:
Tried compiling the default file, and modified for my A10m. Neither will compile. Error no 2. File not found. 

FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\snaps\\Documents\\3D Printer Firmware\\BIGTREETECH-SKR-V1.3-master\\BTT SKR V1.4\\Firmware\\Marlin-2.0.x-SKR-V1.4-Turbo\\.pio\\libdeps\\LPC1769\\_tmp_installing-bv80jbdz-package\\Adafruit_NeoPixel-release\\examples\\StrandtestArduinoBLECallback\\StrandtestArduinoBLECallback.ino':    
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\snaps\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\snaps\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\snaps\.platformio\platforms\nxplpc-arduino-lpc176x\builder\main.py", line 148:
    target_elf = env.BuildProgram()
  File "C:\Users\snaps\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\snaps\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\snaps\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1035:
    project.install_dependencies()
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 867:
    lm.install(uri)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 280:
    force=force,
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 
149:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\snaps\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 
39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\snaps\.platformio\python37\lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\snaps\.platformio\python37\lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target:
============================== [FAILED] Took 1.58 seconds ==============================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       FAILED    00:00:01.580
STM32F103RC                   IGNORED
STM32F103RC_fysetc            IGNORED
STM32F103RC_bigtree           IGNORED
STM32F103RC_bigtree_USB       IGNORED
STM32F103RC_bigtree_512K      IGNORED
STM32F103RC_bigtree_512K_USB  IGNORED
STM32F103RE                   IGNORED
STM32F103RE_bigtree           IGNORED
STM32F103RE_bigtree_USB       IGNORED
STM32F4                       IGNORED
STM32F7                       IGNORED
ARMED                         IGNORED
STM32F103VE_longer            IGNORED
mks_robin                     IGNORED
mks_robin_pro                 IGNORED
mks_robin_lite                IGNORED
mks_robin_lite3               IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FLYF407ZG                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
======================== 1 failed, 0 succeeded in 00:00:01.580 ======================== 
The terminal process terminated with exit code: 1

clfaye, 2020-05-14:
I am getting exactly the same error on an SKR e3 v1.2

Anaferon, 2020-05-22:
Try to put Marlin-2.0.x-SKR-V1.4-Turbo folder to the root of your drive. I was having same problem and that helped.

picard12, 2020-05-22:
Try to rebuild the firmware without clean  the source.

FullTiltt, 2020-06-01:
If the path name is too long it wont compile from what i have experienced ie: 3D Printer Firmware\BIGTREETECH-SKR-V1.3-master\BTT SKR V1.4 i would do as @Mr-ZnO suggested and also change the folder names to something short like Marlin or something.

GregNuspel, 2022-12-17:
I can't seem to get this to build at all I'm getting these errors:

                 from Marlin\src\lcd\lcdprint.cpp:31:
c:\users\greg\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\10.3.1\bits\functional_hash.h:168:3: **error: redefinition of 'struct std::hash<long unsigned int>'       
  168 |   _Cxx_hashtable_define_trivial_hash(unsigned long)**
c:\users\greg\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\10.3.1\bits\functional_hash.h:136:3: note: previous definition of 'struct std::hash<long unsigned int>' 
  136 |   _Cxx_hashtable_define_trivial_hash(wchar_t)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** **[.pio\build\LPC1769\src\src\lcd\lcdprint.cpp.o] Error 1**
================================================================================ [FAILED] Took 42.47 seconds ================================================================================ 

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:42.468

Love to get a solution to this.  Thanks.