bigtreetech/BIGTREETECH-SKR-V1.3 #313: unable to print with skr1.3

fnhhnxfn, 2020-05-03:
hi guys i recently bought 5 x skr1.3 from amazon to upgrade my printer . i wired it them up successfully but i can get it to print 
1) i tried to print via onboard SD it worked for an hour and then stopped at random position and started rebooting itself
2) then i tried to print from my computer and same problem appears the board started to reboot it self
3) then i connected mks tft 35 and tried to print and the same issue again 

i tried printing different gcode , sliced on different slicers updated my board to latest marlin but i cannot get rid of this issue on all of my boards

Pixel-Soup, 2020-05-15:
Which power supply are you using?  Sounds like it's overheating.

fnhhnxfn, 2020-05-17:
Thanks for your suggestion but I don't think that's the issue bcz I am using this power supply for a year with my stock creality board and its being working fine, any suggestions?????

Pixel-Soup, 2020-05-17:
What is the specs on the power supply?  What AWG is the power wires?  
If you provide more details on what you're doing, I might be able to help.  Posting some photos might be useful.

rivendellian, 2020-06-12:
try to change BAUDRATE speed to 250000 in marlin and lcd settings both. i had same problem and i did that and now my printer is not stopping..