bigtreetech/BIGTREETECH-SKR-V1.3 #314: SKR 1.4 TURBO and TFT 24 no printer attached message

jollyrob, 2020-05-05:
I keep getting the no printer attached message in touchscreen mode.
I have tried endless combinations of serial port numbers and baud rates.
I still cannot get the screen to communicate with the board in touchscreen mode.
It is fine in Marlin mode.
Yes, all the connections are made EXP1 and EXP2 and the TFT ( reversed for this board as per the pinout diagram).
Please help, I am considering returning the board as maybe faulty.
Thank you

rdignard, 2020-05-11:
Having similar issue with new SKR 1.4 turbo and TFT 3.5, no printer attached. I too am thinking this board is faulty. I was working with the SKR1.1 Pro board but due to an issue with Y endstop I switched over to the SKR 1.4 board. The display was working with the SKR 1.1 , but now with the SKR 1.4 I get no printer attached.

I-Abraxas, 2020-05-11:
What Stepper motor driver sticks are you using?

jollyrob, 2020-05-12:
I’m using A4988 drivers, are these supported in touchscreen mode? How can I find out? Seems odd not to have them supported.

I-Abraxas, 2020-05-12:
Check the documentation for your specific touchscreen. That should help you find what you are looking for or talk to the manufacturer. I know that my specific touchscreen is compatible with DRV8825, TMC2208, or the TMC2209. I found the beginning of this video helpful when trying to understand what my touchscreen is doing in touchscreen mode: [https://youtu.be/rexqrW6PbnQ](url)

jollyrob, 2020-05-13:
New board and TFT 24 arrived today, and finally the no printer message has disappeared. I used exactly the same connections and firmware as the old board. Hours of time wasted on a defective board!

Ripper80, 2020-08-18:
I have tried two different SKR motherboards, thinking I might have messed one up.  I have not tried replacing the TFT yet....Same issue....endless cables swaps, Marlin changes, but nothing has worked yet,

Eddiiie, 2020-08-19:
Set your baud rate to 115200?

gallanzo, 2020-08-24:
Buongiorno.
Chiedo gentilmente un aiuto.
Ho anch'io lo stesso problema con SKR V4.1 turbo con driver A4988 e TFT 24 V1.1."stampante non collegata".
Ho letto di poli TFT invertiti(ruotando il connettore di 180°?) ed ha funzionato?
Eventualmente si deve sostituire il TFT o la SKR.
Grazie del vostro prezioso aiuto.

labotecno, 2020-09-23:
regulator 3.3 v is dead you need replace it ... you can see the heat with your finger ...

BoomerT6, 2021-05-06:
I have the same problem myself. I have had it connect to the printer in the past, but it was a bit unreliable. After relocating the TFT I just got a blue screen when selecting the Marlin mode. I updated the firmware. Marlin mode now works fine, but now the "No Printer Attached" message has become a permanent feature! 
This thread kinda suggests replacement is the only fix!?