bigtreetech/BIGTREETECH-SKR-V1.3 #316: Connection issue with multiple SKR 1.3/1.4 Boards

Stingaxe, 2020-05-07:
Upgrading my Wanhao I3 board to the SKR. The first SKR 1.3 suddenly lost USB connection to my Octopi while testing the Endstops and I could never get it reconnected to the Pi or back Windows to recognize it. It would see it as Marlin USB device for about 20s but then lose that too. So I figured I just got unlucky with this board. Ordered an SKR 1.4, and it worked fine for about 2 days before the exact same problem happened. This time it happened while moving the Y-axis and I thought maybe I fried the board with back EMF. So I tried one more time with an SKR 1.3, got through 1 print and at the start of the next print lost connection again. 

I'm at a loss at this point. 
     -The SD card is fine, upgraded to a 4 gb and it's worked for 2 of the boards before they stopped 
      working.
     -I've tried switching the serial ports in Marlin, a common suggestion, no change.
   
The printer was working just fine at points, which makes me think it's not an issue with my firmware configuration. Is it possible my power supply or even the Pi is frying the boards?

Thanks for any suggestions!

Marsman1970, 2020-05-08:
One reason could be temperature sensors they will stop the USB from working if there out of range i.e bad sensor or cable.

Stingaxe, 2020-05-09:
> One reason could be temperature sensors they will stop the USB from working if there out of range i.e bad sensor or cable.

Thanks for this tip, I think it put me in the right direction and I've got it working again. The stock firmware that came on the cheapo SD card would actually get the SKR to boot and connect to both the PC and OctoPi. SO I started updating the firmware to see if one of the temp sensors was causing the issue like you said and sure enough temp_sensor_0 was causing the issue.

What I think may have happened is I was using the wrong thermistor type/value for temp sensor 1 and may have burned the 4.7K resistor at the TH0 connection. Could that actually hurt the resistor or could something else have caused it to go?

For the moment I changed the pins in Marlin so that temp_sensor_0 will actually take readings from TH1.

Marsman1970, 2020-05-09:
That resistor should be fine only low voltage on it. But the sensor or wires maybe connectors can go intermittent.Thermistor type/value can definitely cause as i did the same.
Michael

Stingaxe, 2020-05-11:
So far two days of printing no issues with the correct thermistor type! Measuring the resistance of that 4.7K resistor on TH0, which will not work anymore, it shows ~1.3K. The resistors on TH1 and THBed both show 4.7K so I think that the TH0 resistor might actually have burned out. Going to replace it when some 472 resistors come in and see if I can get that connection working again.

Marsman1970, 2020-05-12:
Resistors usually go high not low looks like there is a 10 fu capacitor to ground as well.