bigtreetech/BIGTREETECH-SKR-V1.3 #317: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER dont show something readable

thorstensd, 2020-05-08:
Hello,

I´m having an issue with my Ender 4 Creality Display REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

Backlight is on, but I cant read something. Sometimes, I dont know why, Ascii letters come up. I had also one time the situation, that it shows me the printer screen with temperatures but everything else than numbers was again strange ascii letters. I tested also laguage test, but it dont show something.
Is there an hardware issue known problem? 
I was using marling bugfix version 2.0.x

Thank you.

juanpa669, 2020-05-31:
Hi,
Did you solve your issue ?

I use REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER on SKR 1.4 and it works.
What language did you used ?

Eddiiie, 2020-08-19:
swap the cables?   maybe rotate them too?