bigtreetech/BIGTREETECH-SKR-V1.3 #320: Filament Width Sensor

REDxFROG, 2020-05-10:
In Marlin Configuration_advanced it says

>  *  34 RAMPS_14    : Analog input 5 on the AUX2 connector
 *  81 PRINTRBOARD : Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 RAMBO       : Analog input 3
<

But SKR 1.3 doesn't have AUX2.
Is it still possible to connect a Filament Width Sensor to the board somehow?
It needs to feed 0-5V to the board while 1,75V means 1,75mm Filament.

goldjunge91, 2020-05-11:
then use aux 1 you have there 2x 5V 2xGND  and the tx and rx pins i use this pin for high temp pt100

REDxFROG, 2020-05-11:
hello, thanks.

I have found this in pins_RAMPS.h

```#ifndef` FILWIDTH_PIN
  #define FILWIDTH_PIN                         5  // Analog Input on AUX2
#endif````

of course the same cannot be found in pins_BIGTREE_SKR_V1.3.h

What does 5 mean? Pin 5 ? Does that mean there are schematic with every pin listed as a number?
Sorry I'm completely clueless but I've ordered 2 filament sensors and didn't think it would cause a problem to connect them. If anyone could help me, highly appreciated. Otherwise I have to buy RAMPS board. (which would cause even more problems I fear)

goldjunge91, 2020-05-11:
so make this in the pins_BT_SKR_common.h 
#ifndef FILWIDTH_PIN
#define FILWIDTH_PIN P0_03_A6 // Analog Input on AUX1 RX0 pin 
#endif

[image]

goldjunge91, 2020-05-11:
and way you check the ramps pinout when you have a skr board
maybe u can only this 
 #define FILWIDTH_PIN P0_03_A6

REDxFROG, 2020-05-23:
I'm using the P0_03_A6 (RX0) now but the numbers on the printer display are jumping like no good.
On the multimeter I can get a good 1.74V reading for example.

Even when I plug an empty cable to RX0 or TX0 the numbers are jumping, or when my fingers get close to the cable. Something is way too oversensitive? I don't understand any of this to be honest.

It reads something like $:3.3 : 100% -   $:3.7 : 100% on the display when no cable is connected. Numbers jumping around.

So far I haven't tried to print because I have to add a 3,3V Zener Diode.

PS
Always reads 100%. Even when I set delay to 0cm and error margin  to 2.1mm. But perhaps the calculation only works when it's running.

REDxFROG, 2020-05-23:
Ok for some reason I get wrong readings
Spammed M407 quickly

SENT: M407
READ: Filament dia (measured mm):42.74
READ: ok
SENT: M407
READ: Filament dia (measured mm):42.84
READ: ok
SENT: M407
READ: Filament dia (measured mm):42.76
READ: ok
SENT: M407
READ: Filament dia (measured mm):42.46
READ: ok
SENT: M407
READ: Filament dia (measured mm):42.13
READ: ok
SENT: M105
SENT: M407
READ: Filament dia (measured mm):41.87
READ: ok
SENT: M407
READ: Filament dia (measured mm):41.77
READ: ok
SENT: M105

40mm ? o_O

REDxFROG, 2020-05-23:
I have made a bit of progress,

but the input is floating up and down. I have 2x SKR1.3 , one SKR1.3 has even bigger problems to hold this input number

READ: Filament dia (measured mm):1.69
READ: ok
SENT: M105
READ: ok T:26.02 /0.00 B:26.02 /0.00 @:0 B@:0
SENT: M407
READ: Filament dia (measured mm):1.70
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.70
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.71
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.72
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.73
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.74
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.75
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.76
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.77
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.77
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.78
READ: ok
SENT: M105
READ: ok T:26.14 /0.00 B:26.02 /0.00 @:0 B@:0
SENT: M407
READ: Filament dia (measured mm):1.79
READ: ok
SENT: M407
READ: Filament dia (measured mm):1.80

spammed  "Send M407" again.
It slowly goes up and down and up and down...
The multimeter shows steady 1,76V for example.

This is my math in filwidth.h

  // Convert raw measurement to mm
  static inline float raw_to_mm(const uint16_t v) { return v * 0.2058f * RECIPROCAL(float(MAX_RAW_THERMISTOR_VALUE)); }
  static inline float raw_to_mm() { return raw_to_mm(raw); }

0.2058 instead of 5.0

fabio-bianc, 2020-11-10:
Hi. I had the same problem. Did you find a solution?

goldjunge91, 2020-11-11:
(sorry i didn't see your answers) wich display are you using ? when you use the tft touchscreen then its come from touchscreen i think. but im not 100% sure. i use PIN P0_03_A6 for a e3d pt100 amplifier board and i get never problems with my temps

Manu512, 2020-11-30:
Hi, Same problem.

My math in filwidth.h

  // Convert raw measurement to mm
  static inline float raw_to_mm(const uint16_t v) { return v * 0.126025f * RECIPROCAL(float(MAX_RAW_THERMISTOR_VALUE)); }
  static inline float raw_to_mm() { return raw_to_mm(raw); }