bigtreetech/BIGTREETECH-SKR-V1.3 #321: Smoothieware Laser Engraver

jr-11, 2020-05-11:
I am trying to set up a laser engraver with the smoothieware firmware and config files provided on the skr v1.3 Github page. I have enabled the laser module and set the pins for enabling the laser along with the PWM pin, and when I run the laser I don't get any output on either pin. I'm not quite sure what the problem is. Help would be much appreciated.

derbengale, 2020-07-10:
Im having the same problem here. This is what I added to the configuration file: 

`## Laser module configuration
laser_module_enable                          true            # Whether to activate the laser module at all. All configuration is
                                                              # ignored if false.
laser_module_pin                             3.26             # this pin will be PWMed to control the laser. Only P2.0 - P2.5, P1.18, P1.20, P1.21, P1.23, P1.24, P1.26, P3.25, P3.26
                                                              # can be used since laser requires hardware PWM
laser_module_maximum_power                   .8             # this is the maximum duty cycle that will be applied to the laser
laser_module_minimum_power                   0.0             # This is a value just below the minimum duty cycle that keeps the laser
                                                              # active without actually burning.
laser_module_default_power                   0.3             # This is the default laser power that will be used for cuts if a power has not been specified.  The value is a scale between
                                                              # the maximum and minimum power levels specified above
laser_module_pwm_period                      20              # this sets the pwm frequency as the period in microseconds

switch.laserfire.enable                      true 		#

switch.laserfire.output_pin                  2.0!^			# connect to laser PSU fire (!^ if to active low, !v if to active high)

switch.laserfire.output_type                 digital 		#

switch.laserfire.input_on_command            M3 		# fire laser

switch.laserfire.input_off_command           M5 # laser off`

I tried to set switch.laserfire.output_pin to the pins 3.25 3.26 and 2.0 but neither worked for me. Couldn't test the pwm because I can't switch the laser on and off using M3 or M5 or fire 10/fire off

Is the cnc module necessary for the laser module to work?

dangerousderek, 2020-09-21:
Hey Sorry to bother you, I am also trying to get my K40 running on a skr 1.3. I appreciate the share of the config file, but if you would be willing to share how you wired it up I would be grateful. I am not sure what you mean by cheap level transistor module, so any links and/or pics would be awesome!
thanks!