bigtreetech/BIGTREETECH-SKR-V1.3 #324: BTT SKR V1.3 Motherboard Communication Problems

antonrr90, 2020-05-16:
I thought initially it was a TFT problem, but the odds of 3 different TFTs failing to work with the SKR V1.3 board is highly unusual. And from what I have found on the forums, between the BTT SKRV1.3/1.4 & Several TFTs Produced By BTT are not looking good.

PROBLEM: BTT TFT35 V3.0 would work, but would just randomly stop printing, but would leave the heat on both the bed and extruder running. So I just tried two other TFTs (MKS TFT28 & TFT35), which would not connect at all. You could play with the touch screen menus, but the motherboard wouldn't do anything.

The board is currently running Marlin 2.0, and will connect to my PC using pronterface, but other then that, not a single LCD or Control Interface has come close to working like it should. I may have to return everything as defective at this rate.

antonrr90, 2020-05-16:
Do you realize that RXD0 And TXD0 Are Not Defined In The Marlin pins_BTT_SKR_V1_3.h Nor In pins_BTT_SKR.h? So how is AUX-1 Or The TFT Ports suppose to even work?

antonrr90, 2020-05-16:
By any chance, do you know what the default SERIAL_PORT is for the AUX-1 port is? Currently the only port working is the USB Type B Port which can reliably connect to my PC.

antonrr90, 2020-05-16:
Has anyone looked into "mmu2.h" settings for serial connections? Mine is showing default of MMU_RX_SIZE of 16 instead of the 1024 that has been recommended.

rivendellian, 2020-06-12:
i had same problem with tft24. i connected it to mks gen l v2.0 with touchscreen cable to the aux port of the board.  i fix it with change baudrate. so try to change boudrate settings to 250000 in marlin firmware and lcd settings both. i hope it could help u.

antonrr90, 2020-06-12:
I returned the board, because even after trying several baud rates, still no-joy getting it to work. And it kept stopping mid-print leaving the elements turned on for both the extruder and the heated bed. I sourcing a higher quality board that actually has direct support. It may cost 3 times as much, but at least it will work out of the box.