bigtreetech/BIGTREETECH-SKR-V1.3 #328: Hopefully lost.

Lurchchaos, 2020-05-19:
I have followed every U-tube teaching video but i still can not successfully compile my first flash for my printer... I have a Ender3 Pro with a BTT skr 1.4 turbo board with the TMC2209 steppers,  BL touch 1.3, a BTT SFS and the BTT TFT35. I keep getting errors on the steppers and some other really odd errors mostly i cis not able to define the steppers. . At this point im asking if any one can compile a marlin flash for me.

anm8tor, 2020-05-23:
@Lurchchaos , have you tried starting with the example config files for the Ender 3 provided in the firmware folder?

minagerges9111, 2020-05-24:
I'm having the same problem with the exactly the same configuration

Lurchchaos, 2020-05-24:
I finally solved my issue by figuring out how to compile. So to resolve I opened the the full directory Marlin-2.0.x-SKR-V1.4-Turbo in pio in the  Import Arduino Project and selected the NXP 1769 board. I then tried to compile the default config which pulled in my libraries and I was off to the races. Once i proved it worked i was able to follow all the  teaching videos and compile. Now may isse seems to be hardware incompatibilities with the screen and the board.