bigtreetech/BIGTREETECH-SKR-V1.3 #333: skr 1.4 turbo issue Ender 3 pro

minagerges9111, 2020-05-23:
Hi 
I have been struggling for a week trying to compile marlin2 bug fix on my new SKr 1.4 turbo, it simply doesn't work 
here are the error codes

Linking .pio\build\LPC1769\firmware.elf
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/sd/SdBaseFile.cpp.o: in function `Sd2Card::writeBlock(unsigned long, unsigned char const*)':
C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference 
to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/sd/SdBaseFile.cpp.o: in function `Sd2Card::readBlock(unsigned long, unsigned char*)':
C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:35: undefined reference 
to `SDIO_ReadBlock(unsigned long, unsigned char*)'
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/sd/SdBaseFile.cpp.o: in function `Sd2Card::writeBlock(unsigned long, unsigned char const*)':
C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference 
to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/sd/SdVolume.cpp.o: in function `Sd2Card::writeBlock(unsigned long, unsigned char const*)':
C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference 
to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:36: undefined reference to `SDIO_WriteBlock(unsigned long, unsigned char const*)'
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/sd/SdVolume.cpp.o: in function `Sd2Card::readBlock(unsigned long, unsigned char*)':
C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:35: undefined reference 
to `SDIO_ReadBlock(unsigned long, unsigned char*)'
c:/users/mina gerges/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/sd/cardreader.cpp.o: in function `Sd2Card::init(unsigned char, unsigned char)':
C:\Users\Mina Gerges\Documents\Ender 3 Pro\Marlin-bugfix-2.0.x\Marlin-bugfix-2.0.x/Marlin\src\sd/Sd2Card_sdio.h:34: undefined reference 
to `SDIO_Init()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1

minagerges9111, 2020-05-24:
note: I'm a new to all of that and I'm trying to find my way through

Splashyn, 2020-05-24:
cheap solution from another noob: https://youtu.be/6uGqpCe3Am0

closed 2020-05-27 by minagerges9111