bigtreetech/BIGTREETECH-SKR-V1.3 #334: Marlin 2.0.5 for skr 1.4 Turbo?

t4rif4ri, 2020-05-26:
Hi guys,

i tried to use the newest marlin bugfix version 020005 for my 1.4 turbo board, but i cant compile it.
I get different pin ord board name errors. The newest marlin doesnt even list the turbo version in boards.h

I have now the 20000 version and wanted to update to 20005 cause of the wifi for the board.

zaforas, 2020-05-26:
Aparently, in that version the board name has changed. in configuration.h try to change it to BOARD_BTT_SKR_V1_4_TURBO:
    ```
//#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4_Turbo
    #define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
```

t4rif4ri, 2020-05-27:
> 
> 
> Aparently, in that version the board name has changed. in configuration.h try to change it to BOARD_BTT_SKR_V1_4_TURBO:
> ```
> //#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4_Turbo
> #define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO

true. my mistake. Just saw the skr 1.4 version above at the 1768 part.