bigtreetech/BIGTREETECH-SKR-V1.3 #335: Why Cut Diag pin for endstop for UART - there is no point on the SKR 1.4

zirrer, 2020-05-26:
ABSOLUTELY NO BTT TMC 2209 V1.2 PINS WERE ALTERED 
TLDR: Using a 3 Wire mechanical Endstop without stallguard you dont need to cut diag pin for endstop function
my issue is that cutting the pin was ever considered an option
if i am buying parts for use i shouldn't have to damage/bend/cut new parts when they were supposed to be made for each other

I spent the last couple days investigating a comment listed in the instructions for SKR 1.4 turbo "Note: When using the TMC2209, if you don’t use the stallguard function,
you need to cut off the DIAG pin on the driver so that the mechanical switch
can work normally" because I'm using this config without modifying the BTT TMC 2209 V1.2 in UART with mechanical switches(3 Wire) and  and it works fine no issue. the endstops trigger just fine and i've been printing this way since feb 2020.

**but** my friend  bought the same SKR 1.4 turbo with the BTT TMC2209 V1.2 set up for UART. mechanical enstops(2 Wire) other than dimensions the enstops being 2 wire was the only difference  

so i converted my Y_Max endstop to a 2 with by removing the 5volt wire (red) from the 4Way jst as shown in picture below

[image]
When i remove the 5 volt i started to get the issue everyone was talking about . i did a M119 with the switch released and pressed (shown in pic below)

[image]

so now i finally reproduced the issue, like the picture above no matter if the endstop was press or released Y_max was in a triggered State. So now with the conditions matched both my friends and to the Noted comment in the SKR instructions about cutting/bending Diag pin 

so i reconnect the 5Volt (red) back to my Y_max Endstop like pic below

[image]

And Ran the same M119 test with the Y_max both released and pressed. like pic show below

[image]

everything is working as it should marlin shows that the endstops are triggering like it should but according the the BTT SKR 1.4 instruction THIS IS NOT POSSIBLE AT ALL

but here is the thing when i built mine i was not aware of cutting this pin and works beautifully  until my friend had issues and had to research this 

so i ask again why was cutting the diag pin Ever considered as an acceptable "fix"

Shamoon78, 2020-07-27:
What you have using 2 wires or 3 wires endstop, adding the 5volt endstop wire or removing it is nothing to do with "diag"pin BTT stated to be cutted from the driver, you can use 2 wires mech endstops by changing the confiuration.h setting in marlin 
#define Y_MIN_ENDSTOP_INVERTING false 
you have to make it false and connect the signal and ground of the endstop on your board to the external sides of the endstop switch leaving the middle one unconnected , in regards cutting the pin , i have gtr board v1.0 and i had to bend the pins for 2209 to make the mechanical endtops to work no matter use 3 wires or 2 wires config. and is mistale by BTT they tried to fix it on gtr boards by including a jumper beside each endstop connecter but they made mistake again and that jumper is wrong designed and is not working, so perhaps is the case for skr1.4 turbo i think they included on every manual to cut the pin for each board to buy themselfs hips of customer support issues which they are not good at it anyway, they tring to fix their mistakes but still few to go, to mention some like, absence of saving rom on skr pro, completly failer screen model "btt tft 3.5 v2" which have issues to connect to any of their boards .
but i appreciate their acheivment for making highend boards and drivers available for low cost and i personal prefer their boards even over duet boards.

pediegoivan, 2020-09-15:
I have been searching everywhere about this.
It seems that DIAG pin pulls low all the time except when a stall is detected. On the other hand, mechanical endstops are usually wired to pull low only when triggered, this wiring opposes what the diag pin does.
If you connect the endstop to +V instead of GND, the DIAG pin will be pulling low until the endstop pulls high and the triggered state is detected, even more, if the DIAG pin pulls high marlin should also detect it, so detecting with and endstop and a stall at the same time should be possible. My only concern is if the DIAG pin of the tmc2209 can handle the 5V signal, and if the pulldown of the output could be damaged by the pulse of the endstop. I will keep searching and post here for whoever comes after me

pediegoivan, 2020-09-15:
I just confirmed. DIAG Pin is a Digital output pin. It pulls low until triggered, then outputs high, giving it 5V does nothing to the pin. All Digital pins are 3.3 to 5V capable, and pull up/down resistors are typically 166 Kohm for the tmc2209. So if we give it 5V when pulled low it draws less than 40uA which is under the 2mA stated as current of the low state.
So, yes. You can wire switches that pull to 5V when triggered in parallel to DIAG pins safely and even use both detections at the same time (maybe for example as xmin stallguard and xmax endstop, or delta probing)

Angelol125, 2020-11-16:
@pediegoivan Sorry for the ping. I'm not experienced with mcu stuff and I don't understand how the endstops should be wired so they work without sniping the diag pin. Can you elaborate a bit more? Thank you in advance!

ViktorFran, 2020-11-16:
@pediegoivan 
I am not quite sure of your exact endstop specifications, but it will usually be labelled 'S, G, V' for each of the pins (if its the three pin endstop.)  And on the skr 1.4 board pins, connect the wires in the order 'S, V, G' with the slit is on the upper side. Description by word is a little hard, but when the switch is open, left and middle pin on the board should be connected, and when it's closed, left and right pins should be connected with middle pin disconnected. (When the connector slit is facing up)

pediegoivan, 2020-11-16:
To avoid cutting the diag pin just wire the endstop to pull high when triggered (Normally open and wired only between S and V)

Angelol125, 2020-11-16:
[image]

like this?

pediegoivan, 2020-11-16:
No idea bud, I always got the microswitches only without any pcb thing. No idea if that's NO/NC and which way

Angelol125, 2020-11-16:
I see, thanks. This is an ender 3 switche btw. Could i ask you to explain with this one instead?

[no pcb limit switch](https://frc-pdr.readthedocs.io/en/latest/_images/limitswitch.png)

ViktorFran, 2020-11-16:
@Angelol125 If that's the exact model of switch you're using, connect G to +5V, and V to GND, despite what it says. If you're only using two cables, just connect S and G to 1.2* and +5V respectively. In other words, connect NO pin to +5V, NC to GND, C to signal switch. (You might need to invert the switch logic in the marlin or something)

Angelol125, 2020-11-16:
Thanks @ViktorFran! Will try it today or tomorrow and report back! I am using klipper btw so I hope this can work there as well. I know how to invert pin logic there so I think it will. Thanks again to both of ya!

pediegoivan, 2020-11-16:
I wouldn't connect to gnd at all. No idea on what resistors does it have to pull down. I would connect just G to +5V and S to S

charlesrg, 2022-05-09:
Trade off between not cutting the diag pin or having normally open endstops, which if you forget to connect and endstop you won't know.

numericOverflow, 2025-03-17:
Just replying so I can find this later.  Great work @Bacon-Fixation

0xCA, 2025-06-27:
@Bacon-Fixation please re-upload the pics (and please never rely on discord for storing anything)