bigtreetech/BIGTREETECH-SKR-V1.3 #338: SKR 1.4 Turbo unable to update firmware

hantrax, 2020-05-27:
Hi..
I have an SKR 1.4 Turbo with detached sd-card connector bought from Aliexpress BTT Global store, I am trying to get it replaced but I don't know if I will succeed.
I have successfully uploaded Marlin directly from VisualStudioCode, but now I can't upload a new firmware... from VSCODE, i receive an error on upload_port.
Any idea?

Thanks
Stefano

Sundancer78, 2020-05-29:
Hello, did you define the upload port in VS?
Why does one take off the SD card connection or the SD reader?
via FTDI adapter there should still be a possibility to flash.

hantrax, 2020-05-29:
Hi..
I received the card with the sd-card reader detached.
I try to define port on VS but I don't work.
Can I use the BTT flash writer to flash?

elmolinger, 2020-06-18:
> Hi..
> I received the card with the sd-card reader detached.
> I try to define port on VS but I don't work.
> Can I use the BTT flash writer to flash?

Did this problem still persist?

hantrax, 2020-06-18:
Hi...
I have bought a new SKR 1,4.. and work fine.
I cant't upload a new firmware on the other board..

orrgry, 2020-09-12:
> Hi...
> I have bought a new SKR 1,4.. and work fine.
> I cant't upload a new firmware on the other board..

Hi, does your new board have any issue when flashing updates of marlin like your old board?