bigtreetech/BIGTREETECH-SKR-V1.3 #339: Thermistor input bad

thekylemars, 2020-05-30:
Board was working fine for about 2 months. After being turned off for about a week and turned back on the temperature for Hotend 0 is showing stuck at 184 degrees. Even after removing everything from the board. From what I've gathered the best solution, besides replacing board, is to remap pins. Do I do this in the pins file or configuration_adv?

JeffMedd, 2020-06-08:
Same thing happened to me. Turns out I had shorted the TH0 socket when I accidentally touched the thermistor wire against the aluminium block with a screwdriver. You can check if this has happened to you by measuring the resistance across the resistors in front of the socket (the upper one will be zero for TH0 and 4.7k ohms for the others if its blown). I thought "no problem, I can just use TH1". No matter what I tried, changing pin settings everywhere, still  could not get it to work, as you say always reading a maxed out temp. Ended up replacing the board. Would be very happy if someone was able to come up with a solution.

thekylemars, 2020-07-01:
Actually, it turns out, that I also moved the pins and thermistor to the second input with same result. One reads 300 deegrees and the other 208. Thanks for the tip, I will try and check the resistance.

I swear I must own 10 of these boards in every imaginable stage of needing repaired. I pulled one brand new out of the box the other day and It won't even power up and it's past my amazon return.
It sucks because when these boards work they are great, but I am getting tired of constantly having some problem or another. I do have one board that's been operating solid for about 5 months now, so I guess I have that. Equally unimpressed with the TFT35 v3 displays.

Jon-Oak, 2020-08-02:
I have the same problem, did you solve it?
In my case the thermistor wire was outside the plastic tube and touched against the aluminium block, I've tried change T0 to T1, but it didn't work. I bought a new board, but I would like to use the old one untill it's dies once and for all.

gdattn, 2023-06-24:
I have successfully solved dead T0 pin problem. In my case voltage on dead MCU pin was 0.5V and 3.3V on T1 and TB. I was  unable to install driver properly, SD-card was not connected over usb. I`ve made following changes in Marlin-bugfix-2.1.x :
1. disabled chamber sensor #define TEMP_SENSOR_CHAMBER 0 (in Configuration.h)
2. redirected pin and commented some lines in \src\pins\lpc1768\pins_BTT_SKR_common.h 
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN                    P0_25_A2  // A1 (T1) - (68) - TEMP_0_PIN
#endif
//#ifndef TEMP_1_PIN
//  #define TEMP_1_PIN                    P0_24_A1  // A2 (T2) - (69) - TEMP_1_PIN
//#endif
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN                  P0_23_A0  // A0 (T0) - (67) - TEMP_BED_PIN
#endif

//#if HOTENDS == 1 && !REDUNDANT_TEMP_MATCH(SOURCE, E1)
//  #if TEMP_SENSOR_PROBE
//    #define TEMP_PROBE_PIN            TEMP_1_PIN
//  #elif TEMP_SENSOR_CHAMBER
//    #define TEMP_CHAMBER_PIN          TEMP_1_PIN
//  #endif
//#endif
3. updated firmware over SD-card