bigtreetech/BIGTREETECH-SKR-V1.3 #340: Neopixel on SKR 1.4

juanpa669, 2020-05-31:
Hi,

I try to make work my Neopixel strip but no way it does not works.

`#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     P1_24       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 30       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  //#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

thanks for your help.`

mitofun, 2020-07-18:
Hi juanpa669 
me too cannot start the neopixel on skr 1.4 turbo - tft 3.5 v2 or 3 (((

CBDesignS, 2020-10-13:
first off try to change the line #define NEOPIXEL_TYPE NEO_GRB to #define NEOPIXEL_TYPE   (NEO_GRB + NEO_KHZ800)
then recompile and upload to board, if that has no joy then you may be suffering the same problems as me and I found by trial and error that the NEOPIXEL_PIN P1_24 does not seem to drive neopixel strings.
It must be bugged somewhere in the code or even maybe a hardware board problem. If you do not use filament detection on e0det or e1det then you can reconfigure them to drive the neopixel string and rewire the 3pin plug as the pinout is different but it should work. 
you need to uncomment the P1_25 or P1_26 in pins_BTT_SKR_V1_4.h
//
// Filament Runout Sensor
//
//#define FIL_RUNOUT_PIN                     P1_26  // E0DET
//#define FIL_RUNOUT2_PIN                    P1_25  // E1DET

then you can change the Neopixel pins in configuration.h
#define NEOPIXEL_TYPE   (NEO_GRB + NEO_KHZ800) // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
 #define NEOPIXEL_PIN     P1_25