bigtreetech/BIGTREETECH-SKR-V1.3 #342: SKR 1.4 Turbo + TMC2209

statiealexandru, 2020-06-03:
Hello,

trying to configure the SKR 1.4 Turbo + TMC2209 setup.

Apparently everything is working OK, yet the motor does not move.

The motor is good, the board is powered (jumper set correctly). 

Any ideas?

Send: M122
Recv: 		X	Y	Z	E
Recv: Address		0	0	0	0
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	795	1436	1436	1436
Recv: MAX current	1121	2025	2025	2025
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual	12/31	0/31	0/31	0/31
Recv: PWM scale	14	0	0	0
Recv: vsense		1=.18	0=.325	0=.325	0=.325
Recv: stealthChop	true	false	false	false
Recv: msteps		16	256	256	256
Recv: tstep		max	0	0	0
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: off time	3	0	0	0
Recv: blank time	24	16	16	16
Recv: hysteresis
Recv:  -end		-1	-3	-3	-3
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stst			*	*	*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:0C:00:00
Recv: 		Y	0x00:00:00:00	 Bad response!
Recv: 		Z	0x00:00:00:00	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... Error: All LOW
Recv: Testing Z connection... Error: All LOW
Recv: Testing E connection... Error: All LOW
Recv: ok

AncientAEON, 2020-06-07:
I'm having this same issue. I had forgotten about the DIAG pin and my ender5 kept halting on me.
i removed them from the x and y drivers and now it seems to think its actually moving but yet doesnt move.  My bltouch even deploys for the bed to raise bed none of my motors are moving.
i thought this was an issue with my config files or maybe something i messed up with in the firmware but i've gone through every line and have triple checked all of my work. haven't found anything that would suggest otherwise. the confusing thing to me is the on my skr v1.3 i needed jumpers all down the left side under the driver and now i only need 1 on the Misco pin.  it really seems like there should be a jumper in there somewhere. it's like they aren't receiving a signal to move. sorry this isn't much help.

statiealexandru, 2020-06-08:
Hey!

I have figured out the cause, but not the solution.

The driver enable pin is always high. It remains 3.3v when it should be enabled. 

I have desoldered the enabled pin and connected it to a ground pin on the driver and the motor moved normally.

However, in search for a permanent fix i have tried editing the marlin code, even setting the pin to low manually and it does not really work. Something is pulling it up.

Out of desperation i have connected the enable pin (still connected) to a ground pin on the mainboard and that resulted in smoke and a fried board. DO NOT do the same mistake. A software fix is needed. I’ve opened the same issue on the Marlin github repo and ordered another board...

statiealexandru, 2020-06-08:
Possible solution i’ve got on another thread: 

You have disabled the dedicated enabled pins.
Please read the following and disable it in your Configuration_adv.h

/**
   * Software enable
   *
   * Use for drivers that do not use a dedicated enable pin, but rather handle the same
   * function through a communication line such as SPI or UART.
   */
  #define SOFTWARE_DRIVER_ENABLE

AncientAEON, since you have a functional board and i don’t have one anymore, could you please test this out?

AncientAEON, 2020-06-08:
Yea I can test it this afternoon when I get off work. Thanks. That line does ring a bell, not sure if it is enabled but I'll definitely look. I also order new 2225's to see if maybe it's the 2209's that are the problem. 

Will update you on the code in about 10hrs 
The drivers will be here tomorrow.

Thanks for the help

statiealexandru, 2020-06-09:
Solved with the provided config

closed 2020-06-09 by statiealexandru