bigtreetech/BIGTREETECH-SKR-V1.3 #343: SKR v1.4 Fan0 (CNC / part cooling fan) Control Issues

nverdonk, 2020-06-03:
I recently installed an skr v1.4 board on my Ender 3 pro.  First install and Marlin boot went well.  I printed a test cube and everything worked as intended.

I then tweaked a few firmware settings such as Probe Z-offset and sensorless homing sensitivity values. (I know this can be done on the printer but I wanted to change my config files so that any new firmware changes I did in the future would have the proper values)

Flashed the new firmware and now my part cooling fan will not work.

Details:
- Fans: 2x 5015 blowers wired in parallel (both run at 12 V; using buck converter to set voltage)
- Using Fan0 pin (2.3)
- Have tried setting fan speed with menu commands, Pronter face commands, and direct G-Code
- Used M43 to test if pin is recognized by firmware (Returns P2.03 is FAN_PIN and protected)
- Have used FAN_SOFT_PWM and altered scaling factor with no results
When I set the fan speed, the main menu screen shows the fan animation moving and correctly shows the value I set the speed to.  I probed the pins on the board and, depending on which combination of FAN_SOFT_PWM and scaling factor I used, there was a constant 5.3V or 2.24V across the pins.  These voltages did not change when I changed the speed the fans were set to.

My hotend fan is configured to come on only at hotend temperatures above 50 oC and it works as intended.  I have also tested the above configurations of part fan settings with the nozzle at 180 oC to make sure there is no limit in the firmware that prevents the part fan from turning on.

If anyone has had a similar problem, or knows a potential fix; it would be appreciated!

ChipTechno, 2020-06-04:
It won't work like that. You should wire the fan in serial and connect directly to the fan port.

nverdonk, 2020-06-04:
Thanks for letting me know. I was reading a lot of conflicting options about wiring in parallel or series, so for the point of trouble shooting, I reconnected the original fan and was still having the same control issues.

ChipTechno, 2020-06-04:
Does you connect it with the buck converter? Buck converter does not work well with input from pwm signal. And can you share your config file?