bigtreetech/BIGTREETECH-SKR-V1.3 #344: SKR 1.4 turbo with TMC5160 all high errors

Mace-maker, 2020-06-06:
Hi,
Just installed my new board and drivers with fresh Marlin 2.0.x bug fix based Kay3D cheetah 5.0 firmware on my Ender 5 plus according to Tripod’s Garage YouTube tutorial with the difference being that Tripod has TMC2208 drivers. I also have the TFT3.5 screen and have flashed the unified firmware on it.
I get all high errors and don’t know what to do about this?

[image]

Mace-maker, 2020-06-07:
Since last post I’ve reflashed with debugging M43 gives the following
[image]
[image]

I’ve soldered clk to ground on one driver and raised the clk pin out of play with precise heating using a soldering iron, no difference....

Mace-maker, 2020-06-09:
No one?

Quinnthouzand, 2020-06-17:
I'm having the same issue ;(

Mace-maker, 2020-06-17:
There’s got to be a way to fix this?!

Mace-maker, 2020-06-25:
Finally, finally, finally.

The BTT pin picture in the manual describes TMC5160 SPI as:
XCS pin 1.10
YCS pin 1.9
ZCS pin 1.8
E0CS pin 1.4
MISO pin 0.5
MOSI pin 1.17
SCK pin 0.4

As a Homo sapiens logicus maximus, for the life of me, I cannot understand that they don’t write what they mean:
XCS pin P1_10
YCS pin 1_09
ZCS pin 1_08
E0 pin 1.04
MISO pin 0.05
MOSI pin 1.17
SCK pin 0.04

No, I won’t accept any flack from electronics nuts that this is just how pins are described in the lingo, complete b.s., point. 1.9 is not 1.09, jeez.

To be clear, you have to:
- Change X_CS_PIN etc line 129 onward in pins_BTT_SKR_V1_4.h
-  Make sure Z_MIN_PROBE is set to P0_10 in all configuration files
- By the way do NOT change the -1 values in configuration_adv.h from line 2270 onward by the way, leave them uncommented as well.

Side note, to upload directly to the board without SD card set line 415 Configuration.h SERIAL_PORT to -1 and SERIAL_PORT_2 to 0 in line 421.

Oh yeah, I also changed the printer name to “Hail Mary, mother of God, please just work!”

closed 2020-08-19 by Mace-maker

tca72, 2021-03-19:
Hello 
I also notice that if you disable (comment) #define TMC_USE_SW_SPI  with TMC5160 the ALL HIGH error occurs.

Mace-maker, 2021-03-19:
> Hello
> I also notice that if you disable (comment) #define TMC_USE_SW_SPI with TMC5160 the ALL HIGH error occurs.

Yes, I've since figured out that you have to set this in configuration_adv.h =>

   * Software option for SPI driven drivers (TMC2130, TMC2160, TMC2660, TMC5130 and TMC5160).
   * The default SW SPI pins are defined the respective pins files,
   * but you can override or define them here.
   */
  #define TMC_USE_SW_SPI
  #define TMC_SW_MOSI       P1_17
  #define TMC_SW_MISO       P0_05
  #define TMC_SW_SCK        P0_04

And then everything will work for BTT SPI drivers, TMC5160 and the TMC2130

Pity they just don't add this to the instructions, this is all you have to do.