bigtreetech/BIGTREETECH-SKR-V1.3 #346: skr 1.4 turbo bootloader

Anion-anion, 2020-06-12:
Hi!
I have a skr 1.3 Board using the Flash Magic program and installing the boot jumper, I was able to flash the bootloader into the Board.
There is also a scr 1.4 turbo Board , but there is no boot jumper on it and flash magic cannot load the Bootloader on it. How do I switch it to boot mode and how do I connect the rx tx on the Board, do I have a diagram, do I have any ideas?

Marsman1970, 2020-06-13:
See the 2 blue points above the CPU that's the pins. Here is the file i uploaded ripped from a Skr 1.4 turbo https://github.com/Marsman1970/SKR-V1.3-Bootloader. The file seems a bit different from a Skr 1.3 but i.m gonna try it on a 1.3 & see what happens[BTT SKR V1.4PIN.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4774730/BTT.SKR.V1.4PIN.pdf)

Anion-anion, 2020-06-14:
> See the 2 blue points above the CPU [BTT SKR V1.4PIN.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4774730/BTT.SKR.V1.4PIN.pdf)
> that's the pins. Here is the file i uploaded ripped from a Skr 1.4 turbo https://github.com/Marsman1970/SKR-V1.3-Bootloader. The file seems a bit different from a Skr 1.3 but i.m gonna try it on a 1.3 & see what happens

Marsman1970, Thank! I used to connect the circuit from skr 1.3 instead of skr 1.4. I will, I hope did not hurt anything

Matty74, 2020-06-14:
I have the same problem with my skrq.4 turbo. How do I connect it to load the bootloader

Anion-anion, 2020-06-18:
Turbo couldn't load the loader on the 1.4 Board.Power is available but there is no connection. I checked everything and connected it to the TFT screen. The power is on, the led lights up, but it doesn't load the bootloader. Who knows what the error might be?

kfrancis, 2020-06-23:
@Marsman1970 So, is that actually a bootloader useful on the 1.4 or is that only for the 1.3?

Marsman1970, 2020-06-24:
https://github.com/Marsman1970/SKR-V1.3-Bootloader
The lpc1769 is from a skr 1.4 turbo not sure if it will work on a standard 1.4  i think it should. The lpc1768 1.3 is from a skr 1.3
Michael

kfrancis, 2020-06-25:
@Marsman1970 Turbo was what I was looking for thanks. Now just to figure out how to hook up this FTDI ..

ardiehl, 2020-11-19:
I have a modified bootloader for the SKR 1.3, 1.4 and a.4Turbo, currently tested on 1.4, will test it on 1.4T and 1.3 over the weekend. This one supports update from the internal SD card, from the SD card on the display board (RepRap, BTT TFT35 and may be others that are connected to SPI 0 as well) and from USB (when pressing the display button)

[https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader](https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader)

I used a stlink-v2 for the upload

oechslein, 2020-11-24:
@ardiehl: Would be great if you could offer a hex file of your bootloader.

radek8, 2020-12-31:
> If you are blind as myself: There is a compiled bin file: https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/DFU-Bootloader.bin
> And it works, thanks a lot @ardiehl!

Hi, please do you have a link to the stlink-V2 software you used to load the bootloader? I burned the processor and I need to replace it.
Thank you.

oechslein, 2020-12-31:
I flashed via UART, I didn't manage to use stlink.

Sineos, 2021-01-04:
Just for future reference: Flashed a SKR 1.4 Turbo with @ardiehl Bootloader, using a USB to TTL adapter:

**Things needed:**
* USB to TTL adapter, e.g. https://www.amazon.com/dp/B075N82CDL
* Some Jumper wires, e.g https://www.amazon.com/dp/B01EV70C78
  * female to female for the USB to TTL
  * female to male to conveniently contact the pin P2.10 to get the board into ISP mode 
* @ardiehl bootloader hex (attached to this post, or build it your self)
* [Flashmagic](https://www.flashmagictool.com/download.html)

**Procedure:**

1. Power the board via USB or 12V / 24V
2. Contact the USB TTL adapter as shown in the picture (cyan arrows)
    * **RX** of the adapter to **TX** on the board
    * **TX** of the adapter to **RX** on the board
    * GND to GND
3. Contact the SMD resistor to GND of the board (dark blue arrows), e.g. middle pin of an endstop (here the female to male jumper cable comes in handy). This will contact the ISP mode pin P2.10
4. Reset the board while holding the contact (Nr. 3)
5. Shortly after releasing the Reset button also remove the ISP connection (Nr. 3)
6. Start Flashmagic
    * Chose the @ardiehl hex file
    * Set the correct COM port for your USB to TTL adapter (look it up in the windows device manager)
    * Set the correct chip type (LPC1769 for SKR 1.4 Turbo)
    * Set Erase to `Sectors used by file`
    * Press start
7. Wait until finished, then reset the board
8. Flash new` firmware.bin`, e.g. via SD card upload or using the new bootloader's direct flash capabilities  

Flashing via STLink v2 did not work for me, neither from linux shell, nor via PlatformIO but the Flashmagic way was surprisingly painless.

Bootloader Hex (unzip to use):
[DFU-Bootloader.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/5766031/DFU-Bootloader.zip)

[image]
[image]

radek8, 2021-01-04:
Thank you very much
As soon as a new processor arrives, I will try your guide.

radek8, 2021-01-04:
Query
4. Reset the board while holding the contact (Nr. 3)
Does the jumper in the picture mean blue?

Sineos, 2021-01-04:
> 4. Reset the board while holding the contact (Nr. 3)
> Does the jumper in the picture mean blue?

Not sure that I understand the question:
- To get the chip into the ISP mode you need to pull the pin P2.10 to GND during reset
- The P2.10 is easily accessible via the bottom of the SMD resistor
- GND is easily accessible via, e.g. the middle endstop pin

[image]

So you need to make this contact between GND and P2.10 and hold it, reset the board and then **quickly after releasing** the reset button **remove** the contact.

BTW, I misplaced the arrow in the post above. Corrected now.

radek8, 2021-01-04:
Thank you
I thought so

GadgetAngel, 2021-01-07:
> I have a modified bootloader for the SKR 1.3, 1.4 and a.4Turbo, currently tested on 1.4, will test it on 1.4T and 1.3 over the weekend. This one supports update from the internal SD card, from the [...]

How do you build the project? I have an SKR E3 Turbo I would like to try your bootloader on with modifications to the BTN_EN pin and the BEEPER_PIN but when I try to use MAKE on the project I get:
```
PS C:\Repo\BTT_SKR_13_14_14T_SD-DFU-Bootloader> make
FIND: Parameter format not correct
FIND: Parameter format not correct
FIND: Parameter format not correct
FIND: Parameter format not correct
make: *** No rule to make target 'build/system_LPC17xx.o', needed by 'build/nxp.ar'.  Stop.
PS C:\Repo\BTT_SKR_13_14_14T_SD-DFU-Bootloader>
```
Can you help me?

Sineos, 2021-01-07:
> How do you build the project?

Didn't try under Windows but in Linux it builds nicely. Probably you could use the Ubuntu that can be integrated into Windows 10 via the MS Store if you do not have a native Linux box.

GadgetAngel, 2021-01-07:
@Sineos Hey, after a couple of hours of working on this I figured out I needed to build the project on Linux. I have a laptop that runs a Linux OS.  I will try using make on my Linux laptop.

I already backed up the bootloader from the SKR E3 TURBO board.  I just now need to compile the change for holding the beeper pin low for the SKR E3 Turbo board.  

I am confident this bootloader will work on the SKR E3 TURBO but like you said need to back up the original bootloader which I have done just incase I end up bricking the SKR E3 TURBO.

When I am done I will have a manual on how to do backup and restore the bootloader and a github repository for backed up bootloaders for the following boards (I plan to save the bootloader for each of these boards):

SKR V1.3
SKR V1.4
SKR V1.4 TURBO
SKR PRO V1.1
SKR PRO V1.2
GTR V1.0
SKR mini E3 V2.0
SKR E3 TURBO
BT002 V1.0

Sineos, 2021-01-07:
If you build on Debian / Ubuntu on a x86 architecture you will need:
`apt install build-essential gcc-arm-none-eabi`
You need to x-compile for arm.

GadgetAngel, 2021-01-07:
> If you build on Debian / Ubuntu on a x86 architecture you will need:
> `apt install build-essential gcc-arm-none-eabi`
> You need to x-compile for arm.

is this the right command:
make ARCH=arm CROSS_COMPILE=gcc-arm-nont-eabi

GadgetAngel, 2021-01-07:
@Sineos I installed build-essential and now it comes up with an error before without the build-essential it came up with only a warning.  I am running LINUX MINT 19.3 on my ASUS laptop.  How do I fix this error:

```
   CC     build/core-cm3.o
/bin/sh: 1: Syntax error: "(" unexpected
Makefile:157 recipe for target 'build/core_cm3.o' failed
make: *** [build/core_cm3.o] Error 2
```

GadgetAngel, 2021-01-07:
@Sineos I must of had a tab in the wrong spot and in a script file those are very hard to find and can do funny things.  So I started from scratch with the clone repository from @ardiehl repository.  The `apt install build-essential gcc-arm-none-eabi` tip I do appreciate.  But I am running not on BUSTER but Linux Mint 19.3.  I already had done the `apt install gcc-arm-none-eabi` with out the build-essential and I was able to compile.  I went ahead and add build-essential to my Linux Mint OS and redownloaded the clone repository.  This time with every change I recompiled to see where my error was.  I never found it.  I did do a reboot maybe that is what fixed the issue.

anyways, I have the recompiled .bin and .hex file for SKR E3 Turbo.  I need to first upload the saved backup copy of the original bootloader then I will try to load this new one and see if it will work on the SKR E3 Turbo board.  I will let you know if it works or not.  Thanks for the help, it is very much appreciated.

GadgetAngel, 2021-01-08:
@Sineos IT WORKS!! The beeper going off continuously is solved and the bootloader still uploads the firmware with my ONBOARD SD card reader for the SKR E3 Turbo.

Thanks for all your help.

If anyone else is trying to solve the issue with the SKR E3 TURBO making noise upon boot from your display screen speaker please look to this github repository for the modified bootloader done by @ardiehl & @triffid & @GadgetAngel 

This repository will also have a manual on how to do this and a copy of original bootloaders for the BTT SKR boards

https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader

KoviRobi, 2021-01-17:
I have had problems flashing via SD, so I discovered how to flash via SWD with an ST-link. I have put my comments [here](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/compare/master...KoviRobi:patch-1).

markbwalker, 2021-01-18:
> I have had problems flashing via SD, so I discovered how to flash via SWD with an ST-link. I have put my comments [here](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/compare/master...KoviRobi:patch-1).

I noticed if I push the microcard in just a bit, it would be recognized.  Easy to do when updating firmware, but obliviously not for printing files (since it would then eject when I let loose).  Board (Mini E3 V2) was new, so I contacted BTT.  Connector must have a pin bent (despite me being very careful w/it).

KoviRobi, 2021-01-18:
> I noticed if I push the microcard in just a bit, it would be recognized. Easy to do when updating firmware, but obliviously not for printing files (since it would then eject when I let loose). Board [...]

Also should have said, I figured out the bootloader on the board will go into DFU mode when you hold the rotary encoder button of the screen down while resetting. Then you can do `dfu-util -D <firmware.bin>`. That way you don't have to fidget with the SD card (for mine the SD card is inaccessible and the plan was to use an extender, not feasible if I have to hold it in).

markbwalker, 2021-01-19:
> Also should have said, I figured out the bootloader on the board will go into DFU mode when you hold the rotary encoder button of the screen down while resetting. Then you can do `dfu-util -D [...]

I had read the SD microcard connector had some issues, so from day one I used an extender (board is only 1 month old, if that). But I was doing some updating and inserted a SD microcard in the actual connector...very gently, I might add.  But that was all she wrote.  So rather than ripping out the the motherboard, I got the TFT35 display and just print from that.

I figured if I got a replacement SKR Mini E3 again, I'd just have the same issue.  So now I print using the TFT thumb drive connector.  Going to order an extender for that.

I do have Pi2 setting around unused, I may eventually play around with adding that, but for now, I'll just tinker getting to know the BTT/BLtouch setup.

ForlornWinter, 2021-01-29:
I used Sineos' method successfully to flash a new bootloader on to my SKR 1.4 Turbo. A stumbling point for future readers is that you really do have to poke that resistor fairly hard to get a good contact, and you might have to install the VCP (Virtual COM Port) drivers for your FTDI board manually. Big thanks to @Sineos for this guide.

jenniferlee1818, 2021-03-02:
@Sineos Thank-you!  I did exactly as your guide instructed for my SKR 1.4 Turbo and it works!

GadgetAngel, 2021-03-02:
Here is a repository for all bootloaders for BIGTREETECH boards, including SKR V1.4 TURBO. This repository also has manuals to explain how to backup or restore your bootloader to the BIGTREETECH board. All the original bootloader that were shipped with the BIGTREETECH boards are saved in this repository(https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders). The manuals are for Windows 10 users and Linux users. A manual explaining @Sineos method is even provided. https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader

yeahwgeni, 2021-07-13:
Guys,
you have done such a good job by explaining so many totally new things to me - thank you for that. In my case, it still does not work to update the firmware. 
Do I get it right - to have it run like described above one of the two conditions has to be fulfilled? 
- the SD Card Slot on the board is not broken and is able to read the SD Card
OR
- a display with the "rotary button" needs to be connected to be able to jump into the DFU Mode to be able to flash via USB.
is that right?
Since I don't have the display available that I can't push any buttons - is it somehow possible to "fake" the "button press" directly on the board?

Would be super happy to get a answer even if it is not possible. 

Thank you.

PS. I am old but very new to that topic I purchased the board some months ago and started to work with it way too late to be able to get it back. I just want to get it run and somehow I am tired of buying all the additional things like jumpers, stepper drivers, cables, and more cables and USB convertors ( things I never had before at home) just to figure out one day that the board is just broken. :)

GadgetAngel, 2021-07-13:
> Guys,
> you have done such a good job by explaining so many totally new things to me - thank you for that. In my case, it still does not work to update the firmware.
> Do I get it right - to have it run [...]

Try to reload the bootloader for the SKR V1.4 Turbo board.  Follow this guide: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/bootloader_bin/backed_up_original_bootloaders/SKR%20V1.4%20Turbo/Guide%20UART%20Flash%20SKR%20V1.3%20or%20SKR%20V1.4%20or%20SKR%20V1.4%20TURBO%20using%20Windows%2010.pdf

KoviRobi, 2021-07-14:
> * a display with the "rotary button" needs to be connected to be able to jump into the DFU Mode to be able to flash via USB.
>       is that right?
>       Since I don't have the display available that [...]

Yes, it is :)

> PS. I am old but very new to that topic I purchased the board some months ago and started to work with it way too late to be able to get it back. I just want to get it run and somehow I am tired of [...]

My approach just requires a USB cable, and lacking the screen, something to connect the two pins the button would do.

So when starting the board, the bootloader checks if a given button is pressed ([dfu button check]), and if so, starts the board in DFU mode ([enter dfu mode]), where you can flash firmware in linux using dfu-util (no idea about Windows, sorry). This button is pin 0.28 ([bootloader dfu pin]), which is set to pull-up so you need to connect it to ground ([dfu is pullup]). Location of pin 0.28 on the SRK 1.3 board on EXP1 [SKR 1.3 pinout] (and 1.4 uses the same wiring for EXP1 and 2 of course).

[dfu button check]: https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/main.c#L251
[enter dfu mode]: https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/main.c#L293
[bootloader dfu pin]: https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/config.h#L55
[dfu is pullup]: https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/main.c#L219
[SKR 1.3 pinout]: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-PIN.pdf

GadgetAngel, 2021-07-14:
> > ```
> > * a display with the "rotary button" needs to be connected to be able to jump into the DFU Mode to be able to flash via USB.
> >   is that right?
> >   Since I don't have the display available that I can't push any buttons - is it somehow possible to "fake" the "button press" directly on the board?
> > ```
> 
> Yes, it is :) See below, but in short, you need to connect pin 0.28 to ground and reset the board, it should enter DFU mode and then you can flash it via USB.
> 
> > PS. I am old but very new to that topic I purchased the board some months ago and started to work with it way too late to be able to get it back. I just want to get it run and somehow I am tired of buying all the additional things like jumpers, stepper drivers, cables, and more cables and USB convertors ( things I never had before at home) just to figure out one day that the board is just broken. :)
> 
> My approach just requires a USB cable, and lacking the screen, something to connect the two pins the button would do.
> 
> So when starting the board, the bootloader checks if a given button is pressed ([dfu button check](https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/main.c#L251)), and if so, starts the board in DFU mode ([enter dfu mode](https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/main.c#L293)), where you can flash firmware in linux using dfu-util (no idea about Windows, sorry). This button is pin 0.28 ([bootloader dfu pin](https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/config.h#L55)), which is set to pull-up so you need to connect it to ground ([dfu is pullup](https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader/blob/main/main.c#L219)). Location of pin 0.28 on the SRK 1.3 board on EXP1 [SKR 1.3 pinout](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-PIN.pdf) (and 1.4 uses the same wiring for EXP1 and 2 of course).

Which motherboard are you working with?

yeahwgeni, 2021-07-14:
I tried that couple of times. But also always trying to check to update the firmware via SD card which does not work AND/OR trying to update the firmware directly via VS Studio connected with USB to the board.

GadgetAngel, 2021-07-14:
Please access the whole page on github to see my reply:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346#issuecomment-880015329

KoviRobi, 2021-07-14:
@GadgetAngel so something you could perhaps clear up for me, I thought the ISP programming was different from the DFU programming, i.e. the ISP makes the LPC1768 enter into the *on-chip* bootloader https://www.waveshare.com/wiki/LPC_ISP_Programming_1, and the DFU programming (the center button on screen while booting) makes the device go into a software bootloader by using the *existing* software bootloader https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader, and only allows you to overwrite the firmware

GadgetAngel, 2021-07-14:
> @GadgetAngel so something you could perhaps clear up for me, I thought the ISP programming was different from the DFU programming, i.e. the ISP makes the LPC1768 enter into the on-chip bootloader https://www.waveshare.com/wiki/LPC_ISP_Programming_1, and the DFU programming (the center button on screen while booting) makes the device go into a software bootloader by using the existing software bootloader https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader, and only allows you to overwrite the firmware
> 
> Edit: I've never done the ISP bootloader bit, only the DFU bootloader one. But I think the ISP programmer does https://www.nxp.com/docs/en/user-guide/UM10360.pdf chapter 32

Please see: https://community.st.com/s/question/0D53W00000cSDz7/the-difference-between-boot-loader-and-dfu-mode