bigtreetech/BIGTREETECH-SKR-V1.3 #348: Skr 1.4 turbo no usb

Marsman1970, 2020-06-13:
I got a new skr 1.4 turbo but the USB didn't work i found a short between the 2 data lines. I de soldered the data lines  of the USB socket & the short went away & it worked. I'm gonna remove the socket to investigate and see why it had a short.

mikejr83, 2020-06-22:
Can you elaborate a little more? I pulled the board out of the box, switched the jumper over to usb from vcc, flashed my configuration of Marlin over and then installed the board in my Ender 3. From that point on I've not had USB or SD card support. I can put firmware on the SD card and it seems to flash to the board, but I cannot connect via Octaprint (Raspberry Pi) or Cura on my Windows 10 machine. It doesn't read anything off the SD card so even putting gcode on the SD card I can hit "Init media" or whatever and it sits for a second and doesn't do anything. Doesn't matter the card or whatever. It's cool that these 2209's and the configuration for BLTouch do something, but without anything to print this board is a paperweight.

pedrodias300398, 2020-08-10:
I am having the same issue, it´s saying that the PC doesn´t recognize the usb , did you solve the problem or not yet?

Marsman1970, 2020-08-10:
Mine is working i didn't get get a notice from this post but i'm gonna try my best to upload a video to explain some things.
Michael

soperskaft, 2020-11-07:
Hei pps. Did u guys figure out what the issue was?