bigtreetech/BIGTREETECH-SKR-V1.3 #349: Retraction issue

sniperxnl, 2020-06-14:
Hello Guys

I have build a new Titan Aero on my Ender.
Using TMC2209 in Uart mode
E silentmode off
Old setup was with bowden and all prints where great, switch to titan aero and Retraction don't work for some reason.
e3d advice max 2mm retraction and speed depends.
My pla is printed on 210, but also tested 205/200 no diffrence.

Any idear what could be wrong, using CURA

[image]
![Uploading WhatsA
https://www.dropbox.com/t/MKP8djNuOKaRPfov config files
pp Image 2020-06-14 at 17.31.43.jpeg…]()

Chris0812, 2020-07-10:
how did the retraction work ? was it a noisy an seemingly ridicolously fast movement? 
COULD someow be related to my issue...
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/372

h0x91b, 2020-07-21:
I am also having TItan (Tevo's replica) and extruder produce screeching sound while printing. 
Then after a few minutes stops moving at all. (I can rotate it using my hands, it is not locked, completely turned off).

Then after stopping print, after ~10 minutes again functional.

Driver TMC2208, potentiometer set to 1.1v (tried 0.85v as well).
SKR 1.4 turbo.

Tried 2 cables and 2 different NEMA 17, result is exactly the same.

Before the upgrade everything was fine. I am sure that this is a software issue.

h0x91b, 2020-07-24:
Changed to:

```
  #define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  // #define STEALTHCHOP_E

  #define HYBRID_THRESHOLD

  #define X_HYBRID_THRESHOLD     150  // [mm/s]
  #define Y_HYBRID_THRESHOLD     150
  #define Z_HYBRID_THRESHOLD       3
  #define E0_HYBRID_THRESHOLD     10

```

Also changed from 16 to 4 micro-steps. 
Now it still makes clicky noise but at least doesn't turned off in the middle of the print.

You probably know it but anyway:

FYI 1
Also spent a lot of time on my stupid error, if EEPROM is enabled then after each upload don't forget to factory reset (M502)

FYI 2
You can change the voltage in the middle of the print.

```
M906 E0 900
```