bigtreetech/BIGTREETECH-SKR-V1.3 #350: [SKR 1.4 TURBO] GPIO connection to RPI

minikN, 2020-06-15:
Hello,

I can't for the life of me figure out what I'm doing wrong connection an RPI to my SKR 1.4 using the GPIO pins. I'm using this picture from the manual as a reference: 

[image],

A pin layout for the RPI can be found [here](https://pinout.xyz/).

I connected it like this:
| RPI | SKR 1.4 |
|-|-|
| 6 (Ground) | GND (TFT) |
| 7 BCM 14 (TXD) | RXD (TFT) |
| 8 BCM 15 (RXD) | TXD (TFT) |

I'm pretty sure I wired it up correctly. I did the following changes to the RPI:

In `/boot/config.txt` I added:
```
enable_uart=1
dtoverlay=pi3-miniuart-bt
```

In `/boot/cmdline.txt` I removed:
```
console=serial0,115200
```

In addition I did:
```
sudo systemctl disable serial-getty@ttyS0.service
```
I then restarted the PI.
In Marlin I changed the following:

In `Configuration.h`:
```
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 0

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
```

I then added the serial port `/dev/ttyAMA0` to OctoPrint:
[image]

Here comes the weird part: When I did this for the first time I was able to connect to my printer via OctoPrint (just once). I thought I would try to change the baudrate to `250000` just to see if it would work. I changed it in Marlin and in OctoPrint accordingly. It didn't work (for whatever reason). Then I changed it back to `115200`. Besides that I didn't touch anything else. However ever since that single successfull connection it won't connect again no matter how I tinker with it. I don't think this is an issie with the RPI but with the printer. This is the terminal output when tryint to connect:

```
Connecting to: /dev/ttyAMA0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x67c792f0, open=True>(port='/dev/ttyAMA0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
```

I also tried re-initializing the EEPROM with a `M502` followed by a `M500` (after switching back to USB connection).

Can anyone help me with this?

Tomblarom, 2020-06-21:
Since you're using a Raspberry, why don't you switch to Klipper? It's working perfectly fine, you're able to change the config in OctoKlipper and you're able to use the processing power of your Pi! :)

I know it's not answering your question, but I don't get it, why people still use Marlin :D

minikN, 2020-06-21:
> Since you're using a Raspberry, why don't you switch to Klipper? It's working perfectly fine, you're able to change the config in OctoKlipper and you're able to use the processing power of your Pi! :)
> 
> I know it's not answering your question, but I don't get it, why people still use Marlin :D

Im currently not interested in looking into Klipper. Im new to 3d printing and I'd lile to fully understand my setup before switching to something new.

zonalimitatore, 2020-07-02:
But the pin you write are correct? pin8 tx0 pin10 rx0, not 7 and 8

WhatTakWhat, 2021-01-10:
> Since you're using a Raspberry, why don't you switch to Klipper? It's working perfectly fine, you're able to change the config in OctoKlipper and you're able to use the processing power of your Pi! :)
> 
> I know it's not answering your question, but I don't get it, why people still use Marlin :D

xD
For me Klipper doesn't allow to use usb for connection RPI 4 to skr 1.4 turbo (btw octopi can see same skr plate with Marlin frome same RPI 4).
I want to use Klipper, but I can't. If you know how to change bin compiler in Klipper (to make him work correctly) - please help.
Even lsusb can't show me any changes on dev directory when I use Klipper firmware on SKR 1.4 Turbo. Maybe you know other workaround?

So I use Marlin because it works. Not because I want to.

closed 2021-01-16 by minikN

heatvent, 2021-01-24:
Hi, suffering through the same issue.  How was this resolved?

WhatTakWhat, 2021-03-16:
> Hi, suffering through the same issue.  How was this resolved?

Hello! In my case: 

re-install Marlin;
Apply base config;
Install Klipper;
Follow Klipper instructions.

It seems like Marlin have more power in firmware, even if Klipper installed.