bigtreetech/BIGTREETECH-SKR-V1.3 #352: TMC2130 V1.2 stepper drivers not working SKR 1.3 and TMC2130 V1.2

klayus, 2020-06-16:
I installed the latest marling bugfix 2.0.x firmware, made all the necesary modifications to config and config_adv, just testing the board with the steppers, nothing else. sending m122 gives this response:
SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	795	795	795	795
MAX current	1121	1121	1121	1121
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual	31/31	31/31	31/31	31/31
PWM scale	255	255	255	255
vsense		1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true
msteps		0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295
PWM thresh.	0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	true	true	true	true
triggered
 OTP		false	false	false	false
off time	15	15	15	15
blank time	54	54	54	54
hysteresis
 -end		12	12	12	12
 -start		8	8	8	8
Stallguard thrs	8	8	8	0
DRVSTATUS	X	Y	Z	E
sg_result	1023	1023	1023	1023
stallguard	*	*	*	*
fsactive	*	*	*	*
stst
olb		*	*	*	*
ola		*	*	*	*
s2gb		*	*	*	*
s2ga		*	*	*	*
otpw		*	*	*	*
ot		*	*	*	*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		Z	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing E connection... Error: All HIGH

i can't move the steppers, nothing happens. the stepper don't hold their positions. Help!

Chris0812, 2020-07-10:
Did you already resolve it? if not: what drivers do you use? fysetc, watterott, BTT, all require some hardware modifications to work properly

mpapam, 2020-11-15:
I am facing the same problem, can  chris0812 eplaaitthe hardware modifications for fysetc 2130s?