bigtreetech/BIGTREETECH-SKR-V1.3 #355: SKR V1.4 Turbo Issues with BL Touch

Arflognir, 2020-06-17:
Hi there,

first of all I don't know that much about programming Marlin, so most of my knowledge is gathered together by YT videos and forum entries.

Since I had huge problems with my capacitive z probe, I decided to use a probe similiar to a BL touch (but from a cheaper supplier). Right befor I configured my firmware, I was using the standard endstop switch, which worked fine.

I'm trying to use the servo and probe port designed to be used for bl touch sensors and made the following changes in configuration.h:

[image]

Right now my printer spits out the message: "Err: EEPROM CRC".

Does anyone have a clue what went wrong?

elmolinger, 2020-06-17:
> Right now my printer spits out the message: "Err: EEPROM CRC".
When does this message appear? At flashing or if the Probe Deploys, if touching...?
My guess, it appear at flashing the firmware on the SKR.

On Line 1102 you did #define y_max_pos y_bed_size-14 <- remove the -14 and try again.

What is the Date of this Marlin Version? It looks like it's 3Month old. Maybe try a newer Version.

Did you Check the Wire connection?
[image]

Arflognir, 2020-06-18:
The error is shown right after flashing.
It's Marlin 2.0.5 and the sensor is wired correctly.

I tried removing the "-14" in line 1102 without any changes.

elmolinger, 2020-06-19:
In marlin mode on the LCD you can confirm or reset these EEPROM Messages. 

Can you upload your configuration.h and configuration_adv for compare?

Arflognir, 2020-06-26:
Sorry that it took so long but the Marlin.zip contains my configuration.h and configuration_adv.h
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4836181/Marlin.zip)