bigtreetech/BIGTREETECH-SKR-V1.3 #356: SKR 1.4 Turbo with TMC2209 Imediate fail on G28

MetalBeard72, 2020-06-17:
I upgraded my V-King with a SKR v1.4 Turbo and TMC2209 drivers. I am using Octoprint. It has been running very well. However, when I went to run a G28 the other day, it immediately failed: " Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Printer halted. kill() called! "
Now, the X and Y will only move slightly. Changing the distance does not effect it either way. The Z axis no longer responds at all. The extruder still works. I have switched out motors and Drivers with known working items. No change. I have not modified the firmware (Marlin 2X). It was working fine. I have checked all my wiring with a meter.
I'm not sure what else to check...
Here are the M122 results prior to engaging the x and y:
Recv:  X Y Z E
Recv: Address 0 0 0 0
Recv: Enabled false false false false
Recv: Set current 900 900 600 900
Recv: RMS current 887 887 581 887
Recv: MAX current 1251 1251 819 1251
Recv: Run current 28/31 28/31 18/31 28/31
Recv: Hold current 14/31 14/31 9/31 14/31
Recv: CS actual 14/31 14/31 9/31 14/31
Recv: PWM scale 16 16 11 16
Recv: vsense 1=.18 1=.18 1=.18 1=.18
Recv: stealthChop true true false false
Recv: msteps 16 16 16 16
Recv: tstep max max max max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn false false false false
Recv: off time 4 4 4 4
Recv: blank time 24 24 24 24
Recv: hysteresis
Recv:  -end 2 2 2 2
Recv:  -start 1 1 1 1
Recv: Stallguard thrs 0 0 0 0
Recv: DRVSTATUS X Y Z E
Recv: sg_result 0 0 0 0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv:  X 0xC0:0E:00:00
Recv:  Y 0xC0:0E:00:00
Recv:  Z 0x80:09:00:00
Recv:  E 0x80:0E:00:00
Recv:
Recv:
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: ok

MetalBeard72, 2020-06-18:
After verifying that the Extruder driver and motor still work. I checked each motor with the E axis to confirm the motors opperate correctly, which they do. I then used the E axis driver on each axis to check if the problem is with the Drivers. I also used each Driver for the E axis. all Drivers worked normally in the E axis but in the x,y, and z the behavior mentioned above still persists.

elmolinger, 2020-06-18:
> After verifying that the Extruder driver and motor still work. I checked each motor with the E axis to confirm the motors opperate correctly, which they do. I then used the E axis driver on each [...]

Can you please upload your configuration.h and configuration_adv.h? I would compare it with mine.

Best regards

MetalBeard72, 2020-06-18:
[ConfigFiles.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4799324/ConfigFiles.zip)

elmolinger, 2020-06-18:
Compare to the Vking settings mentioned on https://vkingprinter.com/knowledge-base/v-king-control/ the config looks ok. But unfornately a marlin configuration example is missing on the vking Github Page.

The current on X & Y Steppers isnt a bit high 1200mA ?
Which Stepper Motors are in use? Did you check the temperature last time they are running?
Where did you get these values? 

Could you test other steppers motors on the X & Y Driver?
Before testing new motors lower your current to 500mA for X & Y Stepper Driver.
If the motors are running ok. If not, raise the current in 50mA steps till they are.

Best regards.

MetalBeard72, 2020-06-18:
Yes, I actually just changed the current per someone elses recommendation. I was running them at 900 when it was working. The motors never seamed overly warm. But I will drop to 500 and test again.

elmolinger, 2020-06-18:
the current depends on the Stepper, 500mA is just a guess. if you have the same steppers for z on x & y Axis so 500mA is the right choice to start with.

MetalBeard72, 2020-06-19:
I changed the current to 500mA. No change. I ordered a new board today.