bigtreetech/BIGTREETECH-SKR-V1.3 #358: [BUG] BTT SKR 1.4 Turbo, extruder stops mid print, TMC2208

Jeroentjj, 2020-06-19:
I installed the BTT SKR 1.4 Turbo and have been able to print for a while. Recently the extruder would stop mid print. X, Y and Z would keep going and heating was kept on. After stopping print extruder wouldn't move with loading/unloading commands, above Cold Extrusion temperature. Rebooting made extruder move again. Changed Vref to 1.1V on all BTT TMC2208 V3.0 stepper drivers and test-printed again, problem remained. After running M122 with Pronterface I discovered UART communication errors on multiple axii. All communications on X, Y and Z are OK now, but I am unable to resolve E0. I also tried changing the pins of E0 to E1, but problem persists there too. I tried multiple stepper drivers, all pads soldered correctly but still ALL LOW on E0. Also tested with known working ones (from X, Y or Z). 

I added the TMC stepper library below:
`TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/master.zip`

Also tried modified library:
`TMCStepper=https://github.com/sjasonsmith/TMCStepper/archive/stopListeningOnBegin.zip`

Also tried adding SoftwareSerialM but to no avail:
`SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM`

### My Configurations
Printer: Ender 5
Board: BTT SKR 1.4 Turbo with 3.5 TFT
firmware: Marlin-bugfix-2.0.x , downloaded yesterday
Stepper drivers: BTT TMC2208 V3.0

configuration.h and configuration_adv.h
[https://www.dropbox.com/s/mn0xzvlfo088iru/Configuration.zip?dl=0](url)

Full M122 output:
```
SENDING:M122
axis:pwm_scale/curr_scale|flags|warncount
		X	Y	Z	E
Enabled		false	false	false	false
Set current	600	600	600	500
RMS current	581	581	581	883
MAX current	819	819	819	1245
Run current	18/31	18/31	18/31	15/31
Hold current	9/31	9/31	9/31	7/31
CS actual	9/31	9/31	9/31	0/31
PWM scale	11	11	11	0
vsense		1=.18	1=.18	1=.18	0=.325
stealthChop	true	true	true	false
msteps		16	16	16	256
tstep		max	max	max	0
PWM thresh.	65	65	65	0
[mm/s]		152	152	15	-
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
off time	3	3	3	0
blank time	24	24	24	16
hysteresis
 -end		-1	-1	-1	-3
 -start		1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E
sg_result
stst					*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:09:00:00
		Y	0xC0:09:00:00
		Z	0xC0:09:00:00
		E	0x00:00:00:00	 Bad response!
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... Error: All LOW
```

### Steps to Reproduce
1. Build marlin bugfix 2.0.x with configuration.h and configuration_adv.h
2. Install BTT TMC2208 V3.0 on all axii
3. Run M122 with pronterface

**Expected behavior:** [What you expect to happen]
report: 
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

Motor motion on E0 when above cold extrusion prevention, completed print

**Actual behavior:** [What actually happens]
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... Error: All LOW
motor motion above cold extrusion prevention, but stops mid print.

#### Additional Information

Issues similar to mine:
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/344](url)
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/75](url)
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/247](url)
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/17](url)

I also followed this instructable:
[https://www.instructables.com/id/TMC2208-UART-on-BigTreeTechBIQU-SKR-V11-and-V13-Co/](url
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4804057/Configuration.zip)

elmolinger, 2020-06-19:
> Changed Vref to 1.1V

Changes was made on the Trimpot?

> I discovered UART communication errors on multiple axii

Wich Axis?

have you tried to raise the current for e0?

Jeroentjj, 2020-06-20:
Hi and thank you for your reply.
Yes the change was made with the trimpot.
I can't quite remember which axis had issues, I believe X, Z and E0. All all working correctly now except E0.
The video in the link below, starting from 0:55, mentions the vref should be between 0.8 and 1.1V for standard nema17 motors. I remember the extruder vref was at 1.4V, and on this vref it shut down mid print. I have not tried raising the vref again. To what voltage would you recommend?

elmolinger, 2020-06-20:
1.4Volt seems too high for me, but shouldn't have killed the driver already.
1.1V is ok (my bad I thought about tmc2209 @ answering)

> Rebooting made extruder move again. Changed Vref to 1.1V on all BTT TMC2208 V3.0 stepper drivers and test-printed again, problem remained. 

M122 shows  Error: All LOW still after the reset on E0? Or OK?

Jeroentjj, 2020-06-20:
M122 keeps giving E0 ALL LOW also after rebooting and also after re-flashing firmware. I have one TMC2130 which I also tried, set up in SPI mode, which resulted in ALL HIGH. The SKR 1.4 Turbo should have internal circuitry to allow direct UART communication with the LPC1769 chip, so I have no UART cable. I did try plugging in a different stepper motor cable, the one from Z axis but had no effect. Could I route the UART signal from the stepper driver to a different pin on the board? Not sure if that might help but could be worth a try.

Jeroentjj, 2020-06-20:
sorry for the broken links, I've updated my replies and fixed the links. I wasn't using the syntax right.

elmolinger, 2020-06-20:
Sry, but routing pins from stepper driver is above my experience.

It looks like broken board. Can you replace it?

Jeroentjj, 2020-06-20:
The board is almost brand new so I don't believe it's broken, however as I've seen so many people with the same or similar problems I feel it is due to how the board is designed. I don't have a replacement by hand, other than the original Melzi board but it is incredibly loud. I'm either considering to go to the original creality silent board I can buy from a local webshop or try out the mks Robin board.

asm7100, 2021-11-04:
I also have this problem, runs a little (Printing) And then it stops. I also uses skr 1.4 turbo and TMC2208 in uart mode.