bigtreetech/BIGTREETECH-SKR-V1.3 #361: SKR 1.4 turbo does not work with TFT35 V3 in marlin mode.

zozo6015, 2020-06-21:
Hello

I have a newly installed SKR1.4 turbo which is connected via two serial cables to the TFT35 V3. Everything works fine until I am trying to get to the marlin mode to get access to some configurations that are not avaible into the touch mode. All I get is black screen which does not respond to the nob motions. The same screen previously worked fine with SKR mini E3 1.2. 

I have in marlin configured both serial ports 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Anyone can help with me fixing this?

Filcanna, 2020-06-30:
Did you try rotating 180 degrees exp1 and exp2 connectors? It's the commonest culprit of 12864 LCD not responding.

ellensp, 2020-07-21:
TFT35 V3 has 3 modes of operation
TFT, 1 rs232 cable lcd tft to controller tft 
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER needs two 2*5 pin cables display exp1 and exp2 connected to controller exp1 and exp2.
CR10_STOCKDISPLAY, display exp3 to controller exp1  (nb sd on lcd does not work in this mode.)

zozo6015, 2020-07-31:
I got the marlin mode somehow working but I only need one cable on exp3 port and the cable has to be disconnected when the printer starts and only connect it once it has started. That way both touch mode and marlin mode works and can switch between the two.

Arganaz, 2020-08-29:
> I got the marlin mode somehow working but I only need one cable on exp3 port 

This means CR10_STOCKDISPLAY is defined, try to disable and just have REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and #define REPRAP_DISCOUNT_SMART_CONTROLLER
 enabled, while using EXP1 and EXP2, you should now have Marlin emulation and touch working, I had the same problem

zozo6015, 2020-08-29:
Great stuff. It works as a charm. Thank you much. This one was one of my long lasting headaches.

closed 2020-08-29 by zozo6015