bigtreetech/BIGTREETECH-SKR-V1.3 #365: SKR 1.3 TMC2208 all drivers "Bad responsec

Ruesti, 2020-06-28:
Hi

I bought 2 Bigtreetech SKR 1.3 with the TMC2208 drivers. The first board had suddenly the error "USB not recognizable". Then I tried the second one. All worked fine but all of my drivers showing "Bad response" when I make a M122. I changed all things in configuration. h and configuration.adv. h. The stepper motors are moving but they don't work correctly. I tried to do a print at some point one of the motors stood still with a loud noise. What can I do?

zonalimitatore, 2020-07-02:
Have you set the jumper on the board?

Ruesti, 2020-07-02:
Yes

zonalimitatore, 2020-07-02:
Driver are already "ready to uart",or you have to do the solder?
Where do you make the M122 command?from pc or screen?
Post a photo both side of one of your 2208