bigtreetech/BIGTREETECH-SKR-V1.3 #366: Print stops randomly

Hopper0168, 2020-06-29:
I have just installed SKR 1.4 turbo with a TFT 35 V2.0, on my Ender 3 Pro, got everything set up correct i think, It prints great until about 65 to 99% and just stops, no message on the screen, bed and E0 are still hot, but no movement. I can reboot the printer and start printing again but it will stop again at a different place.  I have tried a couple of different files, same thing. I did print a small print ( less than 45 minutes ) and it worked good. 
Marlin 2
TFT firmware is updated.
SKR firmware is set for the Ender 3 

Anyone else have this issue?

Hipsta69, 2020-06-29:
I'm Having the same issue on my SKR Mini E3. From the research that I've done so far, the following is recommended:

**Swap this:**

`// @section machine

/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 2

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1`

**For this:**

`// @section machine

/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 2`

Basically swapping the Serial port priority,

Hopper0168, 2020-06-29:
I get 

In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:42,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
In file included from d:\marlin-bugfix-2.0.x\marlin\src\inc\marlinconfig.h:42,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:42,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:42,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:42,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:42,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
In file included from d:\marlin-bugfix-2.0.x\marlin\src\inc\marlinconfig.h:42,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
d:\marlin-bugfix-2.0.x\marlin\src\hal\lpc1768\inc\sanitycheck.h:143:6: error: #error "Serial port pins (2) conflict with E stepper pins!"
  143 |     #error "Serial port pins (2) conflict with E stepper pins!"
      |      ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1

if i do that....

Hipsta69, 2020-06-29:
That's odd, it's referring to pins. however that change has no bearing on the pins used, all it is doing is changing the priority of the 2 ports already used. 

I have just looked at my build for my SKR1.4 and that uses different ports to the mini:

Try

`#define SERIAL_PORT 2

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 0`

Hipsta69, 2020-06-29:
I have just looked at my build for my SKR1.4 and that uses different ports to the mini:

Try

`#define SERIAL_PORT 2

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 0`

Hopper0168, 2020-06-29:
Right now i have the second one commented out and port 1 on ( -1 ) and have one print successful.

Hopper0168, 2020-07-01:
Second and larger print was successful today.

DrTon, 2020-08-04:
I have the same problem: printer stops on long prints at random point. I've checked power line of MCU (3.3v) and I think I found the issue. There are random spikes from time to time, once per 0.5-2 hours, even when printer is doing nothin (all steppers and heaters are off):

[image]

Voltage drops during oscillation from 3.3v to 2.5v (may be even lower), I think that is enough for CPU to stop working normally.
Any ideas what to change in schematic to stop this?

Hopper0168, 2020-08-05:
Perhaps your issue may be the power supply. Since i made the changes above, to the serial port. I haven't had an issue with it stopping.

DrTon, 2020-08-05:
Added capacitor on 3.3v regulator output and didn't seen these oscillations anymore, but behavior is the same, more looks like smoothieware bug (https://github.com/Smoothieware/Smoothieware/issues/1481).

damtrx, 2020-08-19:
I'm into the same issue. Changed my stepper drivers to A4988 and still have the problem. I remember that with 2.0.1 marlin i didn't have this problem so i think it is a firmware issue. Sadly i don't have 2.0.1 firmware anymore

DrTon, 2020-08-20:
I switched from Smoothieware to Marlin and everything works fine.

DarkRedemptor, 2021-03-07:
Thanks Hipsta69,
I think I have the same problem than Hopper0168, andf the trick about the serial port seems to work with my 3dprinter (Tevo Tornado gold with SKR1.4 turbo)

Thanks again :-)

benjmharless, 2024-12-04:
I had a similar issue where when printing from a disk on a BTT TFT35 v3 e3 connected to a SKR1.4 turbo.
I fixed it by enabling RX_BUFFER_SIZE and TX_BUFFER SIZE and setting them to the highest option, and by changing the serial ports to match this: https://user-images.githubusercontent.com/54359396/132397091-d596abcf-750f-422a-bb59-afafc246dc58.jpg
Hope that helps!