bigtreetech/BIGTREETECH-SKR-V1.3 #368: SKR 1.4 board doesn't load firmware or connect to computer.

Rayzena, 2020-07-02:
Just as the title says. When I turn on my SKR 1.4 it turns on the fans and the LCD but nothing shows up. It's just a blank LCD that fully lights up. 

- I cannot connect via my computer or via Octoprint
- On my computer, it doesn't show up in Device Manager or make a noise when it gets connected.
- The firmware.bin doesn't change to a .cur and the default .cur file alone doesn't load anything either
- I have tried multiple sd cards, none have had any different a result
- The SD indicator LED turns on meaning the board detects the SD card's existence

Since it's able to route power to the fans and LCD I can only assume that there is an issue with reading the SD in some way.

rocketkittens, 2020-07-06:
I have a similar issue except my LCD backlight does not light up, perhaps then pins are backwards (180degrees) may need to try cutting off the tabs on the connector so it fits backwards or see if pin remapping is possible...

rocketkittens, 2020-07-06:
I learned how the chip works faster than you so i will share my learnings.

First off all, check your LCD, the connectors are most likely backwards, there's a white tab marking on the chip that indicates the correct orientation of the tabs, in general i believe they are suppose to face downwards.

The solution is to take pilers and slow work the black plastic socket off the pins on the back off your LCD, rotate them 180 degrees and reinsert them on the pins.

Now with LCD working, you will have to pay attention to the VCC/USB jumper, it should be on VCC and that means you will need to connect USB + 12v.

Also in the firmware you have to configure the following setting for your SD card support:
`#define SDCARD_CONNECTION ONBOARD`