bigtreetech/BIGTREETECH-SKR-V1.3 #369: SKR 1.4 Turbo + TMC2209 cant compile marlin using software serial

yuvala4, 2020-07-03:
Hey'
Using vanila Marlin 2.0.X Bugfix - latest, downloaded last week (2.0.6)
Hardware:
- BTT SKR 1.4 Turbo
- 4 * BTT TMC2209 v1.2
- Anet A8 style printer
 
When trying to complie I get the below message.
If I enable the Hardware Serial on the skr board I can complie and the printer works just fine, without uart comunication to the TMC drivers,
```
  //#define X_HARDWARE_SERIAL  Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define E0_HARDWARE_SERIAL Serial1
```
I am using the [0.7.1 TMCStepper from teemuatlut](https://github.com/teemuatlut/TMCStepper), downloaded friectly from the platfromIO

Attached my whole project:
[Marlin Bugfix 2.0.X skr 1.4 turbo.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4868964/Marlin.Bugfix.2.0.X.skr.1.4.turbo.zip)

Thank a lot for everyone who can help,
Yuval

> Marlin\src\module\stepper\trinamic.cpp:57:163: error: no matching function for call to 'TMCMarlin<TMC2209Stepper, 'X', '0', X_AXIS>::TMCMarlin(int, int, float, int)'
   57 | #define TMC_UART_SW_DEFINE(IC, ST, L, AI) TMCMarlin<IC##Stepper, L, AI> stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, float(ST##_RSENSE), ST##_SLAVE_ADDRESS)
      |                                                                                                                                                                   ^
Marlin\src\module\stepper\trinamic.cpp:62:44: note: in expansion of macro 'TMC_UART_SW_DEFINE'
   62 | #define _TMC_UART_DEFINE(SWHW, IC, ST, AI) TMC_UART_##SWHW##_DEFINE(IC, ST, TMC_##ST##_LABEL, AI)
      |                                            ^~~~~~~~~
Marlin\src\module\stepper\trinamic.cpp:63:39: note: in expansion of macro '_TMC_UART_DEFINE'
   63 | #define TMC_UART_DEFINE(SWHW, ST, AI) _TMC_UART_DEFINE(SWHW, ST##_DRIVER_TYPE, ST, AI##_AXIS)
      |                                       ^~~~~~~~~~~~~~~~
Marlin\src\module\stepper\trinamic.cpp:213:7: note: in expansion of macro 'TMC_UART_DEFINE'
  213 |       TMC_UART_DEFINE(SW, X, X);
      |       ^~~~~~~~~~~~~~~
In file included from Marlin\src\module\stepper\trinamic.h:35,
                 from Marlin\src\module\stepper\trinamic.cpp:32:
Marlin\src\module\stepper\../../feature/tmc_util.h:200:5: note: candidate: 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'X'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = X_AXIS; uint16_t = short unsigned int; uint8_t = unsigned char]'
  200 |     TMCMarlin(const uint16_t RX, const uint16_t TX, const float RS, const uint8_t addr, const bool) :
      |     ^~~~~~~~~
Marlin\src\module\stepper\../../feature/tmc_util.h:200:5: note:   candidate expects 5 arguments, 4 provided
Marlin\src\module\stepper\../../feature/tmc_util.h:197:5: note: candidate: 'TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(Stream*, float, uint8_t) [with char AXIS_LETTER = 'X'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = X_AXIS; uint8_t = unsigned char]'
  197 |     TMCMarlin(Stream * SerialPort, const float RS, const uint8_t addr) :
      |     ^~~~~~~~~
Marlin\src\module\stepper\../../feature/tmc_util.h:197:5: note:   candidate expects 3 arguments, 4 provided
Marlin\src\module\stepper\../../feature/tmc_util.h:195:7: note: candidate: 'constexpr TMCMarlin<TMC2209Stepper, 'X', '0', X_AXIS>::TMCMarlin(const TMCMarlin<TMC2209Stepper, 'X', '0', X_AXIS>&)'
  195 | class TMCMarlin<TMC2209Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID> : public TMC2209Stepper, public TMCStorage<AXIS_LETTER, DRIVER_ID> {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\module\stepper\../../feature/tmc_util.h:195:7: note:   candidate expects 1 argument, 4 provided
Marlin\src\module\stepper\../../feature/tmc_util.h:195:7: note: candidate: 'constexpr TMCMarlin<TMC2209Stepper, 'X', '0', X_AXIS>::TMCMarlin(TMCMarlin<TMC2209Stepper, 'X', '0', X_AXIS>&&)'
Marlin\src\module\stepper\../../feature/tmc_util.h:195:7: note:   candidate expects 1 argument, 4 provided

>

zonalimitatore, 2020-07-05:
Why use software serial? I can't compile with that enable

yuvala4, 2020-07-06:
Need it in order to control the TMC2209 thru the board and not thru the jumpers / pot

I downloaded the stable realse of Marling 2.5.3 and it solved the issue

zonalimitatore, 2020-07-06:
Ok nice.
But,im curious.What do you mean with throught the board?If you put the jumper you can control the params in any interface,or lcd,panel.

MadMan23, 2020-10-07:
Hello everyone ! I am new to coding and trying to build a Marlin firmware for Anycubic Predator with SKR v 1.4 TURBO + TMC 2209. I managed to find a good start here https://github.com/halofx/Marlin-2.0-Predator then following other tutorials to adapt the settings to skr 1.4 turbo. All good till the last step, the build ! Everything I change, I overwrite, rewrite doesn't work... I need some help. This is the error I get every time 
[image]

zonalimitatore, 2020-10-07:
Download the motherboard folder of repository,and refresh your