bigtreetech/BIGTREETECH-SKR-V1.3 #378: SKR 1.4 turbo no longer visible to my Mac

edwilliams16, 2020-07-22:
I got my SKR 1.4 Turbo and TFT35 V2.0 this AM and started to install it on my printer.  I setup Marlin 2.0.5 for a minimal configuration (no TFT, BL-Touch or Filament sensor) and successfully uploaded the firmware to the SKR with VSCode.  I then connected up the endstops and checked with M119.  I had to invert them, so I re-compiled, uploaded, retested and all was fine.  I then went through a similar process with the x/y/z steppers.  So far, so good.  Now I connected the extruder stepper, thermistor, heater and the part fan.  I suddenly could no longer connect to the board from my Mac.  I disconnected the printer hardware - still no connection.  In fact, the Mac kernel panicked on some of the attempts. (I tried via Printrun, VSCode and Simplify3D -  ports /dev/cu.usbmodem411 and /dev/tty.usbmodem411 would briefly appear, but I could not connect to them.
   If I recompile the firmware and load it on the SD card via my computer, and insert it in the SKR it does get re-named from firmware.bin to firmware.cur  When I power the board, a single green LED between the SD slot and the USB port lights.
  To see if I could get anything to work via the TFT35 V2 (I bought this by mistake  I wanted V3.0!), I hooked up its serial cable and did a
#define SERIAL_PORT_2 0   
in Marlin and re-loaded the SD-Card.  The TFT35 booted up, but said "No Printer Connected".
  So I'm stuck.  Did my USB UART die on me after maybe 10 minutes of use?  Any ideas to try?

edwilliams16, 2020-08-03:
The reason the Mac would not connect is that TH0 developed a short.  This gave a MAX_TEMP error which shuts the board down and causing an endless boot-loop.

fecetinn, 2021-04-20:
I have the same issue. What did you do next? Change the board? Change the pins? Unfortunately, I have a probe sensor, so changing pins is my last option. Also, interestingly, I had the same issue a while ago, but I thought cables caused that. After playing with cables ( I mean the bed's sensor cable ) a bit, it worked again. But now whether I changed the sensor or play with cable it did not work. What would you recommend?