bigtreetech/BIGTREETECH-SKR-V1.3 #379: Stepper driver Vref not the same after to SKR v1.4 Turbo

CaptainJackas, 2020-07-22:
Some strange problem arised today when I switched my MKS Gen L for a BTT SKR v1.4 Turbo. For my stepper motors (1.7A) in combination with my LV8729 stepper drivers (R100) I need a Vref of about 0.68V (where the max Vref for this stepper is around 0.75V), so I had set the adjustment screw in the correct position for this (did this a few weeks ago on the MKS Gen L board).

However after simply switching the stepper drivers onto the new (SKR v1.4) board the value of Vref is now suddenly 0.48V. I tried to turn the screw to increase the voltage but as it already was almost at its maximum I can't increase the voltage any further than 0.55V which is too low for my motors. Does anyone know what the problem could be here and how to fix this?

Note that I ordered 2 BTT SKR v1.4 Turbo boards and the same occurred on both of them, and I also tried this with 5 different LV8729 stepper drivers without any change in result.

edwilliams16, 2020-07-22:
Is this the 1.44 factor depending on whether you define currents/voltages as rms or amplitude?  There a confusing lack of standardization.

CaptainJackas, 2020-07-23:
UPDATE: So I just tried my DRV8825 stepper drivers and these seem to be working just fine, so it has to be something with the compatibility with the LV8729 drivers as I just checked them on my Gen L board and they're still working fine on that one...