bigtreetech/BIGTREETECH-SKR-V1.3 #381: SKR 1.4 BLTouch

Xlaits, 2020-07-27:


Arakon, 2020-08-17:
Check the wiring. On my BLTouch clone, I had to swap 2 wires for the pinout to match up.

no-trick-pony, 2020-08-18:
With the current main branch of Marlin this was literally everything I had to do to get it working: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/207#issuecomment-675216053 - I did not have to switch a cable for my BL-Touch clone. But someone in that thread reported that there might be a pinout difference for some BL-Touch entities.

h0x91b, 2020-09-25:
@Arakon I feel like I am also having an issue with incorrect wiring, but I am not sure what a proper wire order. I have 3dtouch with green, red and yellow. 

Which wires did you switch?

Arakon, 2020-09-27:
Mine had a blue wire, I think.. so no guarantees.
When holding the connector with the wires up and the smooth side towards you, the right two wires need to be swapped. 
I.e. if the order is green, red, yellow, you need to swap red and yellow.

h0x91b, 2020-09-27:
Thanks, I'll try