bigtreetech/BIGTREETECH-SKR-V1.3 #382: Is it possible to get the shutdown relay and UPS to work?

nesergen, 2020-07-28:
SKR1.4-Turbo
UPS V2.0
Relay v1.2
TFT E3 v3

I bought such a set. Installed, connected, configured Marlin. Added M81 code to the end of the print. After the end of printing, the printer writes "shutting off" (like that), but the electricity continues to flow, the printer is working, the inscription does not disappear. The fans continue to run.  Maybe I misunderstand the principle of its operation and purpose ...?  In addition, it is not clear how to turn the printer back on if the relay turns off the power? 
When I turn off the power (press the on / off button) while printing, the UPS works, the voltage drops for about 10 seconds and the printer turns off smoothly, but the extruder does not rise, but continues to print as long as there is enough power. What could be the problem ?

Did anyone manage to get this things to work?

The connection was carried out according to these instructions:
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf](url)
[https://github.com/bigtreetech/BIGTREETECH-Relay-V1.2/blob/master/BIGTREETECH%20Relay%20V1.2/BIGTREETECH%20Relay%20V1.2/Hardware/%E5%BC%80%E5%85%B3%E5%8A%A0%E6%89%93%E5%AE%8C%E5%85%B3%E6%9C%BA.jpg](url)

Arganaz, 2020-08-29:
double check your UPS pin on board, is on the correct place? and in YOUR marlin config. it sounds like no comms between board and UPS

tiagomaricate, 2022-02-20:
How do you connected and set the pins?

nesergen, 2022-02-24:
@tiagomaricate
It's hard to describe in words. I remember that I did it according to the instructions from the official site. With your attention, I will resume my attempts to tweak these settings. I need extra time. One of these days I will check the hardware and software and tell you about the results.

tiagomaricate, 2022-02-24:
> 

Thank you!
I'm facing some difficulties due all my other pins are in use.
And I believe I'm not understanding the logic behind on how it works.

nesergen, 2022-02-26:
@tiagomaricate 
Good day!
I'm tinkering with the printer. I checked the hardware, my connection is correct. Now I'm checking Marlin's settings. Auto power off seems to work. I'm trying to start print recovery after a power failure.
What interests you: hardware or programming?

nesergen, 2022-02-26:
I can sum up the first results. Printer auto-off works. Print recovery works, but with failures, it needs to additionally deal with the settings and hardware. What do you need help with?

tiagomaricate, 2022-02-26:
@nesergen 
Good day!
Actually both, but more related to the conections.
My config is this way:

PWRDET SKR (signal pin) -----> UPS board (signal pin)
PWRDET SKR (signal pin) -----> RELAY board (power failure detection pin)
????????????                      -----> RELAY board (automatic shutdown pin)

Also, I found these connection diagrams:

[image]

[image]

[image]

Not sure if this is correct. I have only the PWRDET pins free. E1DET, E2DET, and the STOPs pins are in use.

nesergen, 2022-02-26:
Do you have all four modules like mine?
SKR1.4-Turbo
UPS V2.0
Relay v1.2
TFT E3 v3

tiagomaricate, 2022-02-26:
@nesergen 

SKR1.4-Turbo
UPS V2.0
Relay v1.2
TFT70
RRF Wifi
Raspberry Pi 3B

tiagomaricate, 2022-02-26:
Also, two smart filament detectors.

nesergen, 2022-02-26:
We have different screens but I think they are similar in functionality. We can experiment.
I think it`s possible to use any free pins, the main thing is to correctly register them later in Marlin. An example is in the picture. 

There are still free pins on the screen. Is it possible to connect one module to the screen, and the second to the motherboard - I don’t know. Another question, how do you print, via a USB flash drive or from a computer via a USB port? 
If through a USB cable, then I don’t know how print recovery will work. If through a flash card, then which port do you select on the screen or on the motherboard? The screen manufacturer made a very capricious device.

According to this [page](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1193#issuecomment-721913482), you need to use the marlin 2.0.5 version for UPS to work. According to my experiments with the marlin 2.0.7 it`s true.
[image]

nesergen, 2022-02-26:
Personally, I print from a flash card on the motherboard.

nesergen, 2022-03-02:
@tiagomaricate
There is the first successful experiment. Connected PSU to pin P0_10 (name: "Probe" on the image) on the board. Connected POWER_LOSS module to pin P1_00 (name: "PWRDET on the image"). Compiled firmware Marlin-2.0.5.4. In Marlin mode, it was possible to restore the printing from the last point. It was not without jambs, there is something to experiment with.
If you need details - ask.

nesergen, 2022-03-04:
I am glad to report that with Marlin version 2.0.9.3 the power-loss module works fine. In Marlin mode, there were no problems. In touchscreen mode, there is no pop-up message prompting you to resume printing, but if you press the joystick, printing resumes.

tiagomaricate, 2022-03-04:
> I am glad to report that with Marlin version 2.0.9.3 the power-loss module works fine. In Marlin mode, there were no problems. In touchscreen mode, there is no pop-up message prompting you to resume printing, but if you press the joystick, printing resumes.

Great! Thank for the update.
I have a TFT70 and it is connected on EXP1, EXP2 and TFT por on SKR.
Whike SKR have the PWRDET pin, the TFT have PS_ON pin on the back.
I will try to use both. The Probe pin on my SKR is in use.

nesergen, 2022-03-05:
next I will try to use usb on the screen to restore the print.

tiagomaricate, 2022-03-05:
@nesergen 

I did the same as you, but now with the Z-STOP pin.
But I need to finish assembling the top head of my printer to test printing something.
The other thing I did is add an EEPROM module extending the default size.
Do you know any way to add more pins?

tiagomaricate, 2022-03-06:
What did you use in #define POWER_LOSS_STATE? HIGH or LOW?

nesergen, 2022-03-06:
> What did you use in #define POWER_LOSS_STATE? HIGH or LOW?

High

nesergen, 2022-03-06:
> Do you know any way to add more pins?

I don’t know how to add pins, I wasn’t interested in this. I assume it is not possible. I only thought about how to use the most of the pins on the display, and reassign those freed on the motherboard to fit my needs. Looked at manuals and diagrams. A few pins I think can be freed.

Salamonius, 2022-10-12:
> I am glad to report that with Marlin version 2.0.9.3 the power-loss module works fine. In Marlin mode, there were no problems. In touchscreen mode, there is no pop-up message prompting you to resume printing, but if you press the joystick, printing resumes

I cannot get the UPS module to work as intended in touchscreen mode. Pins are defined in marlin, Screen is set to settings detailed in BTT setup guide. Z axis will not raise and hotend and heatbed does not shutoff as intended. How did you get it to work in touchscreen mode?

nesergen, 2022-10-13:
Good day!
  
  I need to clear that I use sd-card on the board. Maybe you use usb-port on the screen. In this case I didn't tested this variant.