bigtreetech/BIGTREETECH-SKR-V1.3 #385: WIFI + SKR 1.4 AND TFT 3.5 E3

renanbarbalho, 2020-07-31:
Hello, I flashed two ESP modules, ESP01 and ESP01S, I tried to put both on my SKR 1.4 board and also on TFT 3.5 E3 V3.0 but I was not successful, the modules do not start and do not show the ESP3D network. I watched several videos to follow the steps and also followed the MTB manual. Can someone help me?

hackra76, 2020-08-10:
You have to compile a new firmware for your ESP01 module and split the partitions as per this video: https://www.youtube.com/watch?v=NGgzw-XayEo

renanbarbalho, 2020-08-10:
@ hackra76 Thanks for your answer! When I tried config I followed all instructions of this video. When I connect my two ESP01 nothing happen. The firmware is correct in hardware. To use ESP01 on SKR 1.4 is necessary DC/DC module?

hackra76, 2020-08-10:
As far as I know, the DCDC module is only needed for high current draw accessories like LEDs etc. I use the esp-01s plugged in TFT35 E3 and it works okay. Does it create a WiFi network if you only power it with 3.3v and GND connected ? Did you switch it to client mode or is it still in AP mode ?

renanbarbalho, 2020-08-10:
Yes, I can navigate em ESP3D without connect on my TFT or my mother board.
Can you share your firmware?

hackra76, 2020-08-10:
I have my printer at work, once I get there tomorrow, I will post it here...

renanbarbalho, 2020-08-10:
@hackra76 thanks.

Another question, your ESP01 is: ESP01 or ESP01-S ?
Are you using Ender 3?

hackra76, 2020-08-10:
It is a Tronxy X3A, SKR 1.4 turbo, TMC2208UART, TFT35-E3 V3.0. The module is ESP-01S. I have it plugged in the TFT35. I had the same problem with no wifi network, until I built the ESP3D base firmware myself with platformio....

pacocatech, 2020-09-04:
@hackra76 can you send a firmware?

hackra76, 2020-09-04:
> @hackra76 can you send a firmware?

It is not that simple, the firmware has to be compiled in Platform.io and uploaded directly to the module together with file system image...