bigtreetech/BIGTREETECH-SKR-V1.3 #386: Can't connect to Bigtreetech SKR V 1.3 Board -- Serial/Com Port not recognized

wprojects, 2020-07-31:
So I was using my board through Pronterface and everything was working fine. All of the sudden I was not able to connect to the board any more. The board still powers on, and my LCD screen still turns on when I power it. I have gone to device manager and tried to check out what is going on. At first when I was running Windows 10, every time I would plug in the printer cable to my computer, it would tell me that the USB device was not recognized and had the yellow Triangle with the ! icon in my device manager. I ended up re installing Windows 10 in hopes of the issue fixing itself. Now, I cannot get it to register in Windows either. I also cannot get the serial port to register in atom or Arduino. It was just working and all the sudden it wasn't. I do not think it is fried, my fuses are in tact still.
I have tried the following things to get it working again

USB driver -- Uninstalling/ Reinstalling/Updating
Switched Pins to USB Mode on board
Reset the board through pronterface
Tried holding the reset button when powering the board on
I also put the firmware.bin file on an sd and tried to boot the firmware manually, but it never accepts the file. I tried to do it on someone else's computer too with no luck and tried a .bin file pre compiled from online too with no luck. 
I have tried Mac/Windows/Linux and I cannot get a com/serial port to register anywhere
I have tried multiple usb to printer cables
I have gone through terminal and tried to check dev ports as well with no luck

I have read online that I might could use my Arduino or RPI to try to burn the bootloader over that way? I read that I could maybe hardwire the rpi to the board via octoprint os and may could get it working that way, but I would still like to be able to connect to my board via usb. 

Questions:
1.Is this a firmware/bootloader issue?
2. What is my best approach to get the board to where the usb is recognized again
My main goal is to get the usb com/serial port to be recognized by my computer again so that I can atleast control the printer through my computer again

edwilliams16, 2020-07-31:
I just bricked my SECOND SKR 1.4 Turbo board with the this same issue.  I load the firmware, do a few tests - works fine.  Then a few boots down the road my Mac no longer sees the board through the USB interface. I've tried many of the things that you have.  I no longer feel the need to try connecting with a Raspberry Pi or a Windows machine, since you've already tried that.  Since the Mac connected fine initially to the second board, it doesn't appear to have been corrupted by the first one. (i.e. the problem is on the SKR)  I  have the same questions as you.  I think I have less than 30 minutes of uptime between the two boards.  But absent a fix, it looks like I'm going to have to find some other board than the SKR.

wprojects, 2020-08-01:
@edwilliams16 I am having the exact same issue with the MKS Gen L V 1.0 after I was messing with the stepper motor drivers on the board. It made a crackling sound, and when I rebooted I could no longer connect to the Serial Port on any OS. The SKR did not have an electrical shorting to my knowledge. The MKS Gen L v 1.0 boots up and powers the lcd screen just like the SKR when powered by USB/PSU. Neither one are recognized as a Serial Port on my computer though..

edwilliams16, 2020-08-01:
So what is the commonality here? 

 I was using Marlin 2.0.x, but as an experiment I tried the 7 month old Marlin on BTT's site. I only changed the TMC2130 to TMC2209 in
#define X_DRIVER_TYPE TMC2209
etc.
I loaded FIRMWARE.BIN on an SD-card and put it in the SKR and booted up.  Still no USB serial port - but as before FIRMWARE.BIN got changed to FIRMWARE.CUR so it appears that the firmware is being read in.  I'm still unclear whether this is a hardware or firmware problem.

wprojects, 2020-08-09:
@edwilliams16 my board isn't even converting the file from .bin to .cur

edwilliams16, 2020-08-09:
@wprojects Seems like my problem is simpler as I only had a communication problem.  The board was booting (into a boot loop!) but wasn't talking to me.

Flapstaart, 2020-09-20:
I have the exact same issue as @edwilliams16: SKR 1.4 turbo, 5xTMC2209, suddenly stopped working, still updating but just not communicating. In Windows I initially see a COM port but nothing seems to be able to connect, 1 or 2 minutes later it disappears and reappears as an unknown USB device.

edwilliams16, 2020-09-20:
@Flapstaart   If it is the same, in my case TH0 had shorted internally.  Marlin interprets this as a thermal runaway on the hotend and shuts down the board, causing an endless boot-loop.  Installing firmware that swaps TH0 and TH1 got the board working again.

Flapstaart, 2020-09-20:
@edwilliams16 Thanks for the tip, pulled out my multi-meter and found that the diode for TH0 is shorted and the resistor is only resisting a third of what it should. Sadly, an updated firmware pointing to the pins for TH1 hasn't brought back the board for me.

hrishivengayil, 2022-02-12:
@edwilliams16 You're a legend. Changed the pins for TH0 and TH1 on pins_BTT_SKR_common.h and now working. Thank you!